syssi/esphome-jk-bms #681: DRY1 and DRY2 current state - JK-PB series - not in protocol or not implemented?

mvincm, 2025-02-14:
Hello,

Is there any way  to read DRY1 and DRY2 current state? It is possible at least on 15.37 firmware version and newest BT android APP > 4.28 version. I could assume that this information is in protocol. It could be very useful. Any idea?

Best regards,
MvincM

syssi, 2025-02-14:
Is the Android App able to indicate the state of the contacts?

mvincm, 2025-02-14:
Yeep. Exactly. But you have to get beta version of android app (btw... still 4.28 version...). But my way of thinking is if app can show status/state so BMS must send this state.

mvincm, 2025-02-15:
Try to test and catch debug on my installation (2xBMS) but ESP32 crash and goes to boot loop. I'm aware of 

"ESP32: Adding all supported sensors can lead to a stack overflow / boot loop. This can be solved by increasing the stack size. (https://github.com/syssi/esphome-jk-bms/issues/63)"

but it seems to work only on arduino framework and not on esp-idf (missing DCONFIG_ARDUINO_LOOP_STACK_SIZE). Do you happened to know how to increase stack size on esp-idf?

mvincm, 2025-02-16:
I've managed to increase stack size on esp-idf. Very rude way to be honest, dirty one. I was trying to use "sdkconfig.defaults" and/or editing yamls file with this:

"platformio_options:
    build_flags: 
      - -DCONFIG_ESP_MAIN_TASK_STACK_SIZE=16384" 

but every time "esphome" when compiling override it with defaults  (3586 bytes). So I found the file with this defaults ("Kconfig" in "C:\Users\user_name\.platformio\packages\framework-espidf\components\esp_system" folder, for Windows of course) and changed there. As I said... very rude way but it works.

I will test it on production next weekend. Meanwhile I will try to save debug and catch DRYs changes on lab environment.

syssi, 2025-02-17:
Which ESP32 model do you use? You are the first one in this situation.

mvincm, 2025-02-17:
ESP32DEV as a model in project and about hardware is something called "ESP-WROOM-32". Maybe it is something different (I'll be sure after tests with two BMSs and bigger stack size) but with one BMS everything is solid and stable, with two of them (both are on 15.38 firmware version) ESP32 crash after max 30 seconds.

syssi, 2025-02-17:
Okay. I wasn't aware of you are talking to two devices using a single ESP32.

mvincm, 2025-02-17:
Assumption: if 3586 bytes is enought for one BMS and all sensors than 16384 bytes should be fine for max 3 BMSs (my to be state).

syssi, 2025-02-18:
```
# dry_1_off.txt
55.AA.EB.90.02.45.32.0E.36.0E.1F.0E.8C.0D.1C.0E.1B.0E.20.0E.E5.0C.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.FF.00.00.00.EA.0D.51.01.01.07.53.00.4F.00.53.00.4F.00.52.00.4F.00.52.00.53.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.F0.00.00.00.00.00.4D.6F.00.00.00.00.00.00.00.00.00.00.D3.00.D7.00.00.00.00.00.00.00.00.63.17.9A.00.00.40.9C.00.00.00.00.00.00.0F.00.00.00.64.00.00.00.DC.4D.00.00.01.01.00.00.00.00.00.00.00.00.00.00.00.00.00.00.FF.00.01.00.00.00.90.03.00.00.00.00.CF.41.3F.40.00.00.00.00.21.0B.00.00.00.01.01.01.00.06.01.00.1C.9C.00.00.00.00.00.00.F0.00.D9.00.D6.00.8D.03.F7.C7.A7.09.54.00.00.00.80.51.01.00.00.01.02.00.00.00.00.00.00.00.00.00.00.FE.FF.7F.DC.2F.01.01.B0.0F.07.00.00.60
55.AA.EB.90.02.4E.32.0E.37.0E.1D.0E.8C.0D.1C.0E.1B.0E.20.0E.E5.0C.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.FF.00.00.00.EA.0D.52.01.01.07.53.00.4F.00.53.00.4F.00.52.00.4F.00.52.00.53.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.F0.00.00.00.00.00.4D.6F.00.00.00.00.00.00.00.00.00.00.D4.00.D7.00.00.00.00.00.00.00.00.63.17.9A.00.00.40.9C.00.00.00.00.00.00.0F.00.00.00.64.00.00.00.E1.4D.00.00.01.01.00.00.00.00.00.00.00.00.00.00.00.00.00.00.FF.00.01.00.00.00.90.03.00.00.00.00.CF.41.3F.40.00.00.00.00.21.0B.00.00.00.01.01.01.00.06.01.00.4E.9C.00.00.00.00.00.00.F0.00.D9.00.D6.00.8D.03.FC.C7.A7.09.54.00.00.00.80.51.01.00.00.01.02.00.00.00.00.00.00.00.00.00.00.FE.FF.7F.DC.2F.01.01.B0.0F.07.00.00.A6
55.AA.EB.90.02.58.31.0E.36.0E.1F.0E.8C.0D.1C.0E.1B.0E.20.0E.E5.0C.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.FF.00.00.00.EA.0D.51.01.01.07.53.00.4F.00.53.00.4F.00.52.00.4F.00.52.00.53.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.F0.00.00.00.00.00.4E.6F.00.00.00.00.00.00.00.00.00.00.D4.00.D7.00.00.00.00.00.00.00.00.63.17.9A.00.00.40.9C.00.00.00.00.00.00.0F.00.00.00.64.00.00.00.E6.4D.00.00.01.01.00.00.00.00.00.00.00.00.00.00.00.00.00.00.FF.00.01.00.00.00.90.03.00.00.00.00.CF.41.3F.40.00.00.00.00.21.0B.00.00.00.01.01.01.00.06.01.00.84.9C.00.00.00.00.00.00.F0.00.D9.00.D6.00.8D.03.01.C8.A7.09.54.00.00.00.80.51.01.00.00.01.02.00.00.00.00.00.00.00.00.00.00.FE.FF.7F.DC.2F.01.01.B0.0F.07.00.00.F1
55.AA.EB.90.02.61.32.0E.37.0E.1D.0E.8C.0D.1C.0E.1B.0E.20.0E.E5.0C.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.FF.00.00.00.EA.0D.52.01.01.07.53.00.4F.00.53.00.4F.00.52.00.4F.00.52.00.53.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.F0.00.00.00.00.00.4E.6F.00.00.00.00.00.00.00.00.00.00.D4.00.D7.00.00.00.00.00.00.00.00.63.17.9A.00.00.40.9C.00.00.00.00.00.00.0F.00.00.00.64.00.00.00.EB.4D.00.00.01.01.00.00.00.00.00.00.00.00.00.00.00.00.00.00.FF.00.01.00.00.00.90.03.00.00.00.00.CF.41.3F.40.00.00.00.00.21.0B.00.00.00.01.01.01.00.06.01.00.B5.9C.00.00.00.00.00.00.F0.00.D9.00.D7.00.8D.03.06.C8.A7.09.54.00.00.00.80.51.01.00.00.01.02.00.00.00.00.00.00.00.00.00.00.FE.FF.7F.DC.2F.01.01.B0.0F.07.00.00.37
55.AA.EB.90.02.6B.32.0E.36.0E.1F.0E.8C.0D.1C.0E.1B.0E.20.0E.E5.0C.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.FF.00.00.00.EA.0D.51.01.01.07.53.00.4F.00.53.00.4F.00.52.00.4F.00.52.00.53.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.F0.00.00.00.00.00.4D.6F.00.00.00.00.00.00.00.00.00.00.D4.00.D7.00.00.00.00.00.00.00.00.63.17.9A.00.00.40.9C.00.00.00.00.00.00.0F.00.00.00.64.00.00.00.F1.4D.00.00.01.01.00.00.00.00.00.00.00.00.00.00.00.00.00.00.FF.00.01.00.00.00.90.03.00.00.00.00.CF.41.3F.40.00.00.00.00.21.0B.00.00.00.01.01.01.00.06.01.00.EC.9C.00.00.00.00.00.00.F0.00.D9.00.D7.00.8D.03.0C.C8.A7.09.54.00.00.00.80.51.01.00.00.01.02.00.00.00.00.00.00.00.00.00.00.FE.FF.7F.DC.2F.01.01.B0.0F.07.00.00.83
```

```
# dry_1_on.txt
55.AA.EB.90.01.BF.36.10.00.00.04.0B.00.00.22.0B.00.00.68.10.00.00.4A.10.00.00.0A.00.00.00.54.10.00.00.18.0B.00.00.5E.10.00.00.04.10.00.00.F0.0A.00.00.40.9C.00.00.03.00.00.00.3C.00.00.00.40.0D.03.00.2C.01.00.00.3C.00.00.00.05.00.00.00.D0.07.00.00.BC.02.00.00.58.02.00.00.BC.02.00.00.58.02.00.00.14.00.00.00.46.00.00.00.20.03.00.00.BC.02.00.00.08.00.00.00.01.00.00.00.01.00.00.00.00.00.00.00.40.9C.00.00.DC.05.00.00.B8.0B.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.01.00.00.00.00.00.00.00.60.E3.16.00.10.32.3C.32.18.FE.FF.FF.FF.9F.E9.1D.02.00.00.00.00.04

[... cut ...] 0.00.00.00.00.00.00.00.00.00.F0.00.00.00.00.00.55.6F.00.00.00.00.00.00.00.00.00.00.B3.00.D5.00.00.00.00.00.00.00.00.63.18.9A.00.00.40.9C.00.00.00.00.00.00.0E.00.00.00.64.00.00.00.A3.4C.00.00.01.01.00.00.00.00.00.00.00.00.00.00.00.00.00.00.FF.00.01.00.00.00.90.03.00.00.00.00.CF.41.3F.40.00.00.00.00.22.0B.00.00.00.01.01.01.00.06.01.00.E1.8F.00.00.00.00.00.00.F0.00.D7.00.D5.00.8D.03.BE.C6.A7.09.54.00.00.00.80.51.01.00.00.01.02.00.00.00.00.04.00.00.00.00.00.FE.FF.7F.DC.2F.01.01.B0.0F.07.00.00.87
```

syssi, 2025-02-18:
```
     0  55 55 55 55 55 55 55 55
     1  AA AA AA AA AA AA AA AA
     2  EB EB EB EB EB EB EB EB
     3  90 90 90 90 90 90 90 90
     4  02 02 02 02 02 02 02 02
     5  61 6B F2 FC 34 3E 32 3C
     6  32 32 30 30 33 35 32 32
     7  0E 0E 0E 0E 0E 0E 0E 0E
     8  37 36 35 35 38 37 37 37
     9  0E 0E 0E 0E 0E 0E 0E 0E
    10  1D 1F 1C 1C 20 20 1F 1F
    11  0E 0E 0E 0E 0E 0E 0E 0E
    12  8C 8C 8A 8A 8D 8D 8C 8D
    13  0D 0D 0D 0D 0D 0D 0D 0D
    14  1C 1C 1A 1A 1D 1D 1C 1C
    15  0E 0E 0E 0E 0E 0E 0E 0E
    16  1B 1B 1A 18 1C 1C 1C 1B
    17  0E 0E 0E 0E 0E 0E 0E 0E
    18  20 20 1F 1F 21 21 20 20
    19  0E 0E 0E 0E 0E 0E 0E 0E
    20  E5 E5 DD DC EF F0 EB EB
    21  0C 0C 0C 0C 0C 0C 0C 0C
    22  00 00 00 00 00 00 00 00
    23  00 00 00 00 00 00 00 00
    24  00 00 00 00 00 00 00 00
    25  00 00 00 00 00 00 00 00
    26  00 00 00 00 00 00 00 00
    27  00 00 00 00 00 00 00 00
    28  00 00 00 00 00 00 00 00
    29  00 00 00 00 00 00 00 00
    30  00 00 00 00 00 00 00 00
    31  00 00 00 00 00 00 00 00
    32  00 00 00 00 00 00 00 00
    33  00 00 00 00 00 00 00 00
    34  00 00 00 00 00 00 00 00
    35  00 00 00 00 00 00 00 00
    36  00 00 00 00 00 00 00 00
    37  00 00 00 00 00 00 00 00
    38  00 00 00 00 00 00 00 00
    39  00 00 00 00 00 00 00 00
    40  00 00 00 00 00 00 00 00
    41  00 00 00 00 00 00 00 00
    42  00 00 00 00 00 00 00 00
    43  00 00 00 00 00 00 00 00
    44  00 00 00 00 00 00 00 00
    45  00 00 00 00 00 00 00 00
    46  00 00 00 00 00 00 00 00
    47  00 00 00 00 00 00 00 00
    48  00 00 00 00 00 00 00 00
    49  00 00 00 00 00 00 00 00
    50  00 00 00 00 00 00 00 00
    51  00 00 00 00 00 00 00 00
    52  00 00 00 00 00 00 00 00
    53  00 00 00 00 00 00 00 00
    54  00 00 00 00 00 00 00 00
    55  00 00 00 00 00 00 00 00
    56  00 00 00 00 00 00 00 00
    57  00 00 00 00 00 00 00 00
    58  00 00 00 00 00 00 00 00
    59  00 00 00 00 00 00 00 00
    60  00 00 00 00 00 00 00 00
    61  00 00 00 00 00 00 00 00
    62  00 00 00 00 00 00 00 00
    63  00 00 00 00 00 00 00 00
    64  00 00 00 00 00 00 00 00
    65  00 00 00 00 00 00 00 00
    66  00 00 00 00 00 00 00 00
    67  00 00 00 00 00 00 00 00
    68  00 00 00 00 00 00 00 00
[... 214 lines cut ...]
   283  00 00 00 00 00 00 00 00
   284  00 00 00 00 00 00 00 00
   285  00 00 00 00 00 00 00 00
   286  00 00 00 00 00 00 00 00
   287  FE FE FE FE FE FE FE FE
   288  FF FF FF FF FF FF FF FF
   289  7F 7F 7F 7F 7F 7F 7F 7F
   290  DC DC DC DC DC DC DC DC
   291  2F 2F 2F 2F 2F 2F 2F 2F
   292  01 01 01 01 01 01 01 01
   293  01 01 01 01 01 01 01 01
   294  B0 B0 B0 B0 B0 B0 B0 B0
   295  0F 0F 0F 0F 0F 0F 0F 0F
   296  07 07 07 07 07 07 07 07
   297  00 00 00 00 00 00 00 00
   298  00 00 00 00 00 00 00 00
   299  37 83 A1 F6 F6 44 3B 87

```

Column 0: Position
Column 1-2: DRY1 off
Column 3-4: DRY1 on
Column 5-6: DRY2 off
Column 7-8: DRY2 on

mvincm, 2025-02-18:
Nice! We've got it. Now just implementation ;) I will try to find corresponding file is sources.

syssi, 2025-02-18:
This is just the transposed raw data from the logs. The byte positions still need to be identified but we are on a good track! :-)

mvincm, 2025-02-18:
Line 281? 

When dry 1 is off dry 2 is off and opposite.

syssi, 2025-02-18:
Could you make another capture of "DRY1 AND DRY2 turned on" to proof the guess? :-)

mvincm, 2025-02-19:
Yes, of course. 

[dry_1_off_dry_2_off.txt](https://github.com/user-attachments/files/18859538/dry_1_off_dry_2_off.txt)
[dry_1_on_dry_2_on.txt](https://github.com/user-attachments/files/18859539/dry_1_on_dry_2_on.txt)

syssi, 2025-02-19:
```
# dry_1_off_dry_2_off.txt
55.AA.EB.90.02.C1.2F.0E.33.0E.1A.0E.88.0D.17.0E.17.0E.1D.0E.D6.0C.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.FF.00.00.00.E5.0D.5B.01.01.07.56.00.4F.00.54.00.50.00.54.00.50.00.53.00.55.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.C5.00.00.00.00.00.24.6F.00.00.00.00.00.00.00.00.00.00.C5.00.C6.00.00.00.00.00.00.00.00.63.11.9A.00.00.40.9C.00.00.00.00.00.00.01.00.00.00.64.00.00.00.32.54.00.00.01.01.00.00.00.00.00.00.00.00.00.00.00.00.00.00.FF.00.01.00.00.00.90.03.00.00.00.00.CF.41.3F.40.00.00.00.00.1D.0B.00.00.00.01.01.01.00.06.01.00.E0.0C.00.00.00.00.00.00.C5.00.D0.00.C5.00.8D.03.7E.9B.A9.09.56.00.00.00.80.51.01.00.00.01.02.00.00.00.00.00.00.00.00.00.00.FE.FF.7F.DC.2F.01.01.B0.0F.07.00.00.E2
55.AA.EB.90.02.CB.2D.0E.32.0E.1B.0E.88.0D.17.0E.17.0E.1D.0E.D7.0C.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.FF.00.00.00.E5.0D.5B.01.01.07.56.00.4F.00.54.00.50.00.54.00.50.00.53.00.55.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.C5.00.00.00.00.00.27.6F.00.00.00.00.00.00.00.00.00.00.C5.00.C6.00.00.00.00.00.00.00.00.63.11.9A.00.00.40.9C.00.00.00.00.00.00.01.00.00.00.64.00.00.00.38.54.00.00.01.01.00.00.00.00.00.00.00.00.00.00.00.00.00.00.FF.00.01.00.00.00.90.03.00.00.00.00.CF.41.3F.40.00.00.00.00.1D.0B.00.00.00.01.01.01.00.06.01.00.17.0D.00.00.00.00.00.00.C5.00.D0.00.C5.00.8D.03.84.9B.A9.09.56.00.00.00.80.51.01.00.00.01.02.00.00.00.00.00.00.00.00.00.00.FE.FF.7F.DC.2F.01.01.B0.0F.07.00.00.32
55.AA.EB.90.02.D5.2F.0E.32.0E.1A.0E.89.0D.17.0E.16.0E.1C.0E.D7.0C.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.FF.00.00.00.E5.0D.5C.01.01.07.56.00.4F.00.54.00.50.00.54.00.50.00.53.00.55.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.C5.00.00.00.00.00.27.6F.00.00.00.00.00.00.00.00.00.00.C5.00.C6.00.00.00.00.00.00.00.00.63.11.9A.00.00.40.9C.00.00.00.00.00.00.01.00.00.00.64.00.00.00.3D.54.00.00.01.01.00.00.00.00.00.00.00.00.00.00.00.00.00.00.FF.00.01.00.00.00.90.03.00.00.00.00.CF.41.3F.40.00.00.00.00.1D.0B.00.00.00.01.01.01.00.06.01.00.4D.0D.00.00.00.00.00.00.C5.00.D0.00.C5.00.8D.03.89.9B.A9.09.56.00.00.00.80.51.01.00.00.01.02.00.00.00.00.00.00.00.00.00.00.FE.FF.7F.DC.2F.01.01.B0.0F.07.00.00.7D
55.AA.EB.90.02.DF.2F.0E.33.0E.1B.0E.89.0D.17.0E.16.0E.1D.0E.D7.0C.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.FF.00.00.00.E5.0D.5C.01.01.07.56.00.4F.00.54.00.50.00.54.00.50.00.53.00.55.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.C5.00.00.00.00.00.27.6F.00.00.00.00.00.00.00.00.00.00.C5.00.C6.00.00.00.00.00.00.00.00.63.11.9A.00.00.40.9C.00.00.00.00.00.00.01.00.00.00.64.00.00.00.43.54.00.00.01.01.00.00.00.00.00.00.00.00.00.00.00.00.00.00.FF.00.01.00.00.00.90.03.00.00.00.00.CF.41.3F.40.00.00.00.00.1D.0B.00.00.00.01.01.01.00.06.01.00.84.0D.00.00.00.00.00.00.C5.00.D0.00.C5.00.8D.03.8F.9B.A9.09.56.00.00.00.80.51.01.00.00.01.02.00.00.00.00.00.00.00.00.00.00.FE.FF.7F.DC.2F.01.01.B0.0F.07.00.00.CD
55.AA.EB.90.02.E9.2F.0E.32.0E.1A.0E.88.0D.17.0E.16.0E.1C.0E.D6.0C.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.FF.00.00.00.E4.0D.5C.01.01.07.56.00.4F.00.54.00.50.00.54.00.50.00.53.00.55.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.C5.00.00.00.00.00.23.6F.00.00.00.00.00.00.00.00.00.00.C5.00.C6.00.00.00.00.00.00.00.00.63.11.9A.00.00.40.9C.00.00.00.00.00.00.01.00.00.00.64.00.00.00.48.54.00.00.01.01.00.00.00.00.00.00.00.00.00.00.00.00.00.00.FF.00.01.00.00.00.90.03.00.00.00.00.CF.41.3F.40.00.00.00.00.1D.0B.00.00.00.01.01.01.00.06.01.00.BA.0D.00.00.00.00.00.00.C5.00.D0.00.C5.00.8D.03.94.9B.A9.09.56.00.00.00.80.51.01.00.00.01.02.00.00.00.00.00.00.00.00.00.00.FE.FF.7F.DC.2F.01.01.B0.0F.07.00.00.0D
55.AA.EB.90.02.F3.2F.0E.32.0E.1A.0E.89.0D.17.0E.16.0E.1C.0E.D6.0C.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.FF.00.00.00.E4.0D.5C.01.01.07.56.00.4F.00.54.00.50.00.54.00.50.00.53.00.55.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.C5.00.00.00.00.00.23.6F.00.00.00.00.00.00.00.00.00.00.C5.00.C6.00.00.00.00.00.00.00.00.63.11.9A.00.00.40.9C.00.00.00.00.00.00.01.00.00.00.64.00.00.00.4D.54.00.00.01.01.00.00.00.00.00.00.00.00.00.00.00.00.00.00.FF.00.01.00.00.00.90.03.00.00.00.00.CF.41.3F.40.00.00.00.00.1D.0B.00.00.00.01.01.01.00.06.01.00.F1.0D.00.00.00.00.00.00.C5.00.D0.00.C5.00.8D.03.99.9B.A9.09.56.00.00.00.80.51.01.00.00.01.02.00.00.00.00.00.00.00.00.00.00.FE.FF.7F.DC.2F.01.01.B0.0F.07.00.00.59

# dry_1_on_dry_2_on.txt
[... cut ...] 0.00.00.00.00.00.00.00.00.00.CA.00.00.00.00.00.FE.6E.00.00.00.00.00.00.00.00.00.00.CC.00.CE.00.00.00.00.00.00.00.00.63.10.9A.00.00.40.9C.00.00.00.00.00.00.02.00.00.00.64.00.00.00.E1.55.00.00.01.01.00.00.00.00.00.00.00.00.00.00.00.00.00.00.FF.00.01.00.00.00.90.03.00.00.00.00.CF.41.3F.40.00.00.00.00.19.0B.00.00.00.01.01.01.00.06.01.00.B6.1D.00.00.00.00.00.00.CA.00.BF.00.CF.00.8D.03.2D.9D.A9.09.56.00.00.00.80.51.01.00.00.01.03.02.00.00.00.06.00.00.00.00.00.FE.FF.7F.DC.2F.01.01.B0.0F.07.00.00.E4
```

syssi, 2025-02-19:
```
     0  55 55 55 55
     1  AA AA AA AA
     2  EB EB EB EB
     3  90 90 90 90
     4  02 02 02 02
     5  E9 F3 9E A8
     6  2F 2F 2B 2A
     7  0E 0E 0E 0E
     8  32 32 2F 30
     9  0E 0E 0E 0E
    10  1A 1A 17 17
    11  0E 0E 0E 0E
    12  88 89 87 85
    13  0D 0D 0D 0D
    14  17 17 15 15
    15  0E 0E 0E 0E
    16  16 16 14 14
    17  0E 0E 0E 0E
    18  1C 1C 18 1A
    19  0E 0E 0E 0E
    20  D6 D6 C9 C9
    21  0C 0C 0C 0C
    22  00 00 00 00
    23  00 00 00 00
    24  00 00 00 00
    25  00 00 00 00
    26  00 00 00 00
    27  00 00 00 00
    28  00 00 00 00
    29  00 00 00 00
    30  00 00 00 00
    31  00 00 00 00
    32  00 00 00 00
    33  00 00 00 00
    34  00 00 00 00
    35  00 00 00 00
    36  00 00 00 00
    37  00 00 00 00
    38  00 00 00 00
    39  00 00 00 00
    40  00 00 00 00
    41  00 00 00 00
    42  00 00 00 00
    43  00 00 00 00
    44  00 00 00 00
    45  00 00 00 00
    46  00 00 00 00
    47  00 00 00 00
    48  00 00 00 00
    49  00 00 00 00
    50  00 00 00 00
    51  00 00 00 00
    52  00 00 00 00
    53  00 00 00 00
    54  00 00 00 00
    55  00 00 00 00
    56  00 00 00 00
    57  00 00 00 00
    58  00 00 00 00
    59  00 00 00 00
    60  00 00 00 00
    61  00 00 00 00
    62  00 00 00 00
    63  00 00 00 00
    64  00 00 00 00
    65  00 00 00 00
    66  00 00 00 00
    67  00 00 00 00
    68  00 00 00 00
[... 210 lines cut ...]
   279  00 00 00 00
   280  00 00 00 00
   281  00 00 06 06 <-- 
   282  00 00 00 00
   283  00 00 00 00
   284  00 00 00 00
   285  00 00 00 00
   286  00 00 00 00
   287  FE FE FE FE
   288  FF FF FF FF
   289  7F 7F 7F 7F
   290  DC DC DC DC
   291  2F 2F 2F 2F
   292  01 01 01 01
   293  01 01 01 01
   294  B0 B0 B0 B0
   295  0F 0F 0F 0F
   296  07 07 07 07
   297  00 00 00 00
   298  00 00 00 00
   299  0D 59 A1 E4
```

Column 2-3: DRY1 & DRY2 off
Column 4-5: DRY1 & DRY2 on

syssi, 2025-02-19:
```
00 = 0x00 = b0000
02 = 0x02 = b0010
04 = 0x04 = b0100
06 = 0x06 = b0110
              ||
              |DRY1 bit
              DRY2 bit
```

mvincm, 2025-02-19:
Nice one! Could you be so kind and implement? If you don't have any free time please just point me correct file from repo and I'll try to patch myself. Thanks.

syssi, 2025-02-19:
I'm preparing a pull request / feature branch and provide some instructions later the day!

syssi, 2025-02-19:
Please update your YAML configuration and give it a try:

```
substitutions:
  # ...
  external_components_source: github://syssi/esphome-jk-bms@add-dry-contacts
  
binary_sensor:
    # ...
    dry_contact_1:
      name: "${name} dry contact 1"
    dry_contact_2:
      name: "${name} dry contact 2"
```

mvincm, 2025-02-19:
Thx! I will try it tomorrow.

P.s.
Nice, so nice :)

mvincm, 2025-02-20:
Full success! Works like a charm. I could confirm.

Thank you!

closed 2025-02-20 by syssi