syssi/esphome-jk-bms #682: Settings password

m2teusz, 2025-02-16:
Hello,
Is it possible to add the option of entering a different password than the factory one for BMS settings (123456). On the factory BMS it reports an alarm and after changing it I can't change the settings using HA ;-)
b.r.
Mateusz

syssi, 2025-02-16:
Which BMS model (hardware/firmware version) do you own? In the past the password wasn't important because it was compared on client side (Android App) once per established connection only. See https://github.com/syssi/esphome-jk-bms/issues/139

May be this has changed over time. Do you talk to your BMS using BLE or UART-TTL connection?

m2teusz, 2025-02-16:
Vendor ID: JK_PB2A16S15P HW: V15A SW: V15.33 Connecting via BLE

syssi, 2025-02-16:
Do you own an Android device to prepare a btsnoop capture of the BLE traffic between the BMS and the Android app?

Danifont, 2025-12-06:
I'm with the same problem with same JK-BMS and version and it's a problem because I've a pre-mounted box and the led of error does not disapear till I change the default password.