syssi/esphome-jk-bms #686: esp8266 rebooting

Smoeller0, 2025-02-21:
Hello
i try install  ( esphome run esp8266-example.yaml ) on d1 mini
it worked connect to bms, connecto to mqtt, publish correct data. shot time but reboot every second and then stop

[10:01:37][C][esphome.ota:073]: Over-The-Air updates:
[10:01:37][C][esphome.ota:074]:   Address: jk-bms.local:8266
[10:01:37][C][esphome.ota:075]:   Version: 2
[10:01:37][C][safe_mode:018]: Safe Mode:
[10:01:37][C][safe_mode:019]:   Boot considered successful after 60 seconds
[10:01:37][C][safe_mode:021]:   Invoke after 10 boot attempts
[10:01:37][C][safe_mode:022]:   Remain in safe mode for 300 seconds
[10:01:37][W][safe_mode:031]: SAFE MODE IS ACTIVE
[10:02:27][I][safe_mode:041]: Boot seems successful; resetting boot loop counter

what can possible go wrong?

syssi, 2025-02-21:
Please reduce your YAML configuration. It's too big/complex to fit into the memory of the ESP8266. If you have the chance use a ESP32.

Smoeller0, 2025-02-21:
thank you so far. 
i remove some of the sensor data from the yaml. not it is notmore reooting.
but read the right Data only one time per boot. then send a nan and only repeat his log messages .

[13:23:32][W][component:237]: Component jk_modbus took a long time for an operation (75 ms).
[13:23:32][W][component:238]: Components should block for at most 30 ms.
[13:23:32][D][uart_debug:114]: <<< A9:08:AA:00:00:00:78:AB:01:AC:01:AD:03:A4:AE:01:AF:01:B0:00:0A:B1:14:B2:31:32:33:34:35:36:80:00:00:B3:00:B4:4A:4B:2D:42:4D:2A:00:B5:32:35:30:32:B6:00:00:45:38:B7:31:37:48:5F:5F:5F:5F:5F:53:31:37:2E:30:34:31:B8:00:B9:00:00:00:78:BA:4A:4B:4B:A8:6A:05:00:00:00:00:00:35:2D:34:31:31:30:31:33:41:30:34:30:C0:01:00:00:00:00:68:00:00:44:7B
[13:23:37][D][uart_debug:114]: >>> 4E:57:00:13:00:00:00:00:06:02:00:00:00:00:00:00:68:00:00:01:28
[13:23:37][D][uart_debug:114]: <<< A7:55:09:00:00:00:80:10:01:79:18:01:87:33:20:54:03:0E:56:82:A4:2A:0D:D3:06:87:A3:50:68:43:0D:B3:80:00:17:81:00:91:80:D1:0B:89:84:00:00:61:27:98:87:00:00:89:80:80:20:31:08:8B:00:00:8C:00:07:8E:0D:20:8F:08:D0:90:10:68:91:0F:A0:92:00:03:93:0B:04:94:0B:22:95:00:03:96:01:2C:97:00:64:98:01:2C:99:00:28:9A:00:3C:9B:0B:B8:9C:00:0A:9D:01:9E:00:50:9F:00:A3:03:46:A1:00:3C:A2:00:14:A3:00:46:A4:00:46:A5:FF:F6:A6:00:80:DA:EC:A8:FF:F6:6A:48:0D:00:00:78:AB:01:AC
[13:23:37][W][component:237]: Component jk_modbus took a long time for an operation (72 ms).
[13:23:37][W][component:238]: Components should block for at most 30 ms.

meccip, 2025-02-28:
if you use d1 mini you can switch to esp32 lolin s2 mini that is the same size to replace in your project. No need to rewire. i had the same issue with 8266 memory and when i replace with esp32 problems magically dissapear.

syssi, 2025-03-01:
Let's close this issue because its a hardware limitation.

closed 2025-03-01 by syssi