syssi/esphome-jk-bms #688: Reboot on ESP32 - Task watchdog got triggered. The following tasks/users did not reset the watchdog in time

mvincm, 2025-02-21:
Hello,

My ESP32 reboots every few minutes.  I catch logs for it. One more observation. Version github://syssi/esphome-jk-bms@add-dry-contacts is much more stable than current main version (sic!).

Could you take a look? My first looks suggest two kinds of events which lead to reboot. Thanks in advanced.

[bms_reboots_cut.log](https://github.com/user-attachments/files/18916137/bms_reboots_cut.log)
[bms_reboots_full.log](https://github.com/user-attachments/files/18916138/bms_reboots_full.log)

Thanks,
MvincM

syssi, 2025-02-22:
The root cause must be something else because there is no relevant difference between the teo branches: 

https://github.com/syssi/esphome-jk-bms/compare/add-dry-contacts...main

syssi, 2025-02-22:
Could you provide your YAML configuration?

mvincm, 2025-02-22:
Hello. About branches... Yeah I'm pretty aware of no major difference (I've analysed code changes as far a as could) but still regardless of the code branch reboot is happening. From debug code it looks like there are two different situations causes reboot.

My yaml is exactly the same like for tests with dry contacts. In attachment. 

[dry_log.yaml.txt](https://github.com/user-attachments/files/18920135/dry_log.yaml.txt)

syssi, 2025-02-22:
Does the situation improve (may be just a bit) if you remove the `web_server` component?

syssi, 2025-02-22:
If I had to guess it's a timing issue / race condition which will happen at both branches and it's just coincidence you have seen this issue on `main`. Let's try to focus on the specific exceptions:

If you build and flash the node using `esphome run yourconfig.yaml` via a serial connection the stack traces will be decoded automatically. Could you try to provide another log including the decoded stack traces?

mvincm, 2025-02-22:
Thanks! I will try both later on.

mvincm, 2025-02-23:
Ok.. So after some tests (on main branch) I could confirm that build without webserver works stable for 1+ hour. 
To be precise. Build with webserver but without open connection form web browser also works. I caught some logs from serial output after command "esphome run my.yaml". Backtrace is much more richer.  Please take a look if you could. Thanks!

[Windows PowerShell_05.txt](https://github.com/user-attachments/files/18931888/Windows.PowerShell_05.txt)
[Windows PowerShell_06.txt](https://github.com/user-attachments/files/18931889/Windows.PowerShell_06.txt)
[Windows PowerShell_04.txt](https://github.com/user-attachments/files/18931890/Windows.PowerShell_04.txt)

syssi, 2025-02-23:
The exception of [Windows PowerShell_05.txt](https://github.com/user-attachments/files/18931888/Windows.PowerShell_05.txt) can be avoided by chaging the log level from `VERY_VERBOSE` to `DEBUG` or `INFO` to free some more memory and reduce load.

syssi, 2025-02-23:
Same applies for [Windows PowerShell_06.txt](https://github.com/user-attachments/files/18931889/Windows.PowerShell_06.txt).

syssi, 2025-02-23:
And [Windows PowerShell_04.txt](https://github.com/user-attachments/files/18931890/Windows.PowerShell_04.txt) too. In all cases it's a lack of memory.

mvincm, 2025-02-23:
Thx! So I will try with INFO and Web Server On. Than try to retest.

mvincm, 2025-03-09:
Fresh build with current main version.  WebServer off, log level INFO. Works rather stable. Thanks!

closed 2025-03-09 by mvincm