syssi/esphome-jk-bms #690: Missing configuration item: SOC-100% Volt

hopkinslaw, 2025-03-02:
This project allows me to vary Cell OVP and Cell OVPR from HA. But I cannot set "SOC - 100% Volt."

I want to alter my charging profile from Home Assistant. The BMS needs the following numbers be in this order: 

Cell OVP> SOC-100% Volt.>Cell OVPR.

If I set the values low, say 3.40, 3.39,3.8, and then want to raise the levels to, say OVP at 3.65 and OVPR at 3.6. I cant set the value for OVPR because it would be higher than SOC-100%. 

Would it be possible to add this configuration item?

I am using a JK-B2A8S20P BMS. My hardware is new and I use the JK02_32S protocol.

syssi, 2025-03-02:
Take a look at the configuration example: https://github.com/syssi/esphome-jk-bms/blob/main/esp32-ble-v14-example.yaml

Search for "soc100" and add this new number entity to your configuration. Does this resolve your issue?

hopkinslaw, 2025-03-02:
Yes! That does it. Nice. Thank you.

syssi, 2025-03-02:
Great! Let me know if you miss something else!

closed 2025-03-02 by syssi