syssi/esphome-jk-bms #694: Adding support for SmartBat BLE BMS

SigLinJo, 2025-03-09:
Hi!

I have a small sailing boat where I want to remotely monitor the electric system. I have a SunWind LFP 100 Ah battery https://www.sunwind.se/alla-produkter/energi/batteri/batteri-sunwind-lithium-high-power-100ah (mine is gray instead of black, bought a couple of years ago) I have managed to get the battery connected to Homeassistant through this project using esphome ble proxy https://github.com/patman15/BMS_BLE-HA where they have support for my battery through the Offgridtec LiFePo4 Smart Pro: type A & B (show up as SmartBat-A… or SmartBat-B…) Mine is showing up as SmartBat-A.

My problem is that using the esphome ble proxy uses a lot of data between the ESP32 and Homeassistant and I'm running homeassistant on my server at home and want to only run the ESP32 in the boat because of power restrictions (I only have 120 Wp of solar on the boat and it has to run the bilge-pump, refrigeration box and navigation instruments). I have managed to connect the ESP32 either through mqtt or with API through an Wireguard tunnel directly from the ESP because ble-proxy only works with api connection. The api connection unfortunately burns through over 200 MB of data per day making the 4g data plan for the boat to expensive and using more power than I want.

I would prefer to handle the ble connection and sensor configuration locally with esphome and only send data to my homeassistant server through mqtt at much lower intervals. Would it be possible for you to add SmartBat support to one of your esphome BMS projects?

Best regards
Sigfrid from Sweden.

syssi, 2025-03-09:
Let's give it a try but please be warned. You have to do a lot of testing. Do you have access to the battery at the moment?

This is the BMS implementation we would like to port to ESPHome: https://github.com/patman15/BMS_BLE-HA/blob/main/custom_components/bms_ble/plugins/ogt_bms.py

SigLinJo, 2025-03-09:
Thank you for getting back to me! I have the battery at home in my homelab and can do any testing required! That seems to bee the implementation that connects to my battery. It shows up like this: 

[image]

syssi, 2025-03-09:
Do you have a spare ESP32 available and could you flash some YAML configuration I will provide?

SigLinJo, 2025-03-09:
I have several ESP32s and can flash them!

syssi, 2025-03-09:
I will setup a project this evening and will inform you about the next steps.