syssi/esphome-jk-bms #696: [Feature request]JK PB2A16S15P external LCD control

wonsky007, 2025-03-10:
Hi,
I'm looking for a feature to control LCD Always On switch, it's possible via JK BMS app.

Currently, I don't see a switch to control the LCD display. This functionality would allow the display to be turned off, for example, during the night or when nobody is home. Is there a plan to implement this feature? Alternatively, how can I help?

syssi, 2025-03-10:
It's already available (via BLE only): https://github.com/syssi/esphome-jk-bms/blob/main/esp32-ble-v14-example.yaml#L336-L337

wonsky007, 2025-03-10:
Indeed, after switching to BLE communication, the feature is available. For me, this is sufficient.

[image]

syssi, 2025-03-10:
The UART-TTL protocol doesn't support this feature.