syssi/esphome-jk-bms #698: B2A20S150P RS485 on GPS port

uploader18, 2025-03-10:
Hello!

I tested RS485 modbus communication on B2A20S150P and B6A20S150P bms and it works. If you select on bms under UART1 protocol 1(JK RS485 V1.0) and connect UART to RS485 converter to ESP you can read modbus register that are the same as in PB  series. Do not forget to set address. Even addressing works so you can connect multiple bms's on same line and read data from them. I bought 12 of B6A20S150P and tested them.I used original UART to RS485  converter.

Here is my yaml

`esphome:
  name: jk-modbus
  friendly_name: JK-MODBUS

substitutions:
  name: "PACK 1"  

esp8266:
  board: d1_mini

uart:
  tx_pin: D5
  rx_pin: D6
  rx_buffer_size: 384
  baud_rate: 115200
  debug:
    direction: BOTH

# Enable logging
logger:
    level: VERBOSE
    baud_rate: 0

# Enable Home Assistant API
api:

ota:

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Jk-Modbus Fallback Hotspot"

captive_portal:

modbus:
  id: modbus1

modbus_controller:
  - id: jkbms
    address: 0x01   ## address of the Modbus slave device on the bus
    modbus_id: modbus1
    setup_priority: -10
    update_interval: 60s
    command_throttle: 50ms

binary_sensor:
  - platform: modbus_controller
    modbus_controller_id: jkbms
    name: "Charging"
    address: 0x12C0
    register_type: holding
    lambda: |-
      return data[0] == 1;

  - platform: modbus_controller
    modbus_controller_id: jkbms
    name: "Discharging"
    address: 0x12C0
    register_type: holding
    lambda: |-
[... 470 lines cut ...]
    name: "${name} balancing"
    custom_command: [0x00, 0x03, 0x10, 0x78, 0x00, 0x02]
    write_lambda: |-
      payload.push_back(0x00);
      payload.push_back(0x10);
      payload.push_back(0x10);
      payload.push_back(0x78);
      payload.push_back(0x00);
      payload.push_back(0x02);
      payload.push_back(0x04);
      payload.push_back(0x00);
      payload.push_back(0x00);
      payload.push_back(0x00);
      payload.push_back(x);
      return true;
    lambda: |-
      if (data.size() != 4 ) {
        ESP_LOGE("Custom", "Invalid data size %d",data.size());
        return {};
      }
      return (data[3] == 1);

    `

regards Žiga

syssi, 2025-03-11:
Thanks for your feedback and the valuable details!

syssi, 2025-03-11:
Could you provide the hardware and software version of the BMS?

uploader18, 2025-03-11:
[image]

[image]