syssi/esphome-jk-bms #706: 2 inverter BMS master+slave

fastvd, 2025-03-20:
Hello. I upgraded my system and now I have 2 batteries and 2 BMSs respectively. One works as a master (all DIP switches are OFF) and the other as a slave (DIP switch in position 1). The master BMS is connected to the deye inverter via CAN. Everything is classic, everyone sees everyone...
I wanted to get data from each battery separately in HA.
and I can't figure out how to contact the master BMS, which has address 0x00, which cannot be used when configuring modbus_controller ...what am I doing or understanding wrong?
if I switch the master to position 1, and the slave to position 2, then I can read all the data from 2 BMS, but then the inverter sees only one BMS (here which DIP is in position 1)

fastvd, 2025-03-21:
I understand that there can be only 1 master.  but the question still remains - how to get janni from 2 BMS at the same time?

syssi, 2025-03-21:
I must admit I'm not an expert of your BMS model. If you use the windows application which connector do you use? Could you provide a screenshot of the windows application and try to identify the used protocol? I guess it's a RS232 connection and a Seplos like protocol.

fastvd, 2025-03-21:
no, I use a usb-rs485 adapter in the same master bms port as the esp32 board with the rs485-ttl converter...now I'm going down to the basement and taking a picture...

fastvd, 2025-03-21:
[image]

[image]

fastvd, 2025-03-21:
esp32 + rs485-ttl
[image]
usb-ttl for PC
[image]
conection port PC or esp32 on master bms
[image]
conection port PC or esp32 on master bms close-up version
[image]
slave bms
[image]

all bms are JK BMS PB2A16S15P

syssi, 2025-03-21:
Could you provide a screenshot of the `detail log` and/or `logging`? I would like to see the raw traffic somewhere.

Silverwings007, 2025-03-21:
> esp32 + rs485-ttl [image] usb-ttl for PC [image] conection port PC or esp32 on master bms [image] conection port PC or esp32 on master bms close-up version [image] slave bms [image]
> 
> all bms are JK BMS PB2A16S15P

Hey fastvd,

Are you using the canbus or RS485 connection on the JK BMS? Is it possible to keep the active connection on the CANBUS between BMS and inverter as well as interrogating the BMS via RS485?

Thanks.

fastvd, 2025-03-22:
between the inverter and the master bms communication via CAN! 
between the master bms and the slave bms communication via rs485! this can be seen in the photo)
and so, at the moment when the inverter is working with the master bms + slave bms, we can still receive data on the PC via rs485

Silverwings007, 2025-03-22:
> between the inverter and the master bms communication via CAN! between the master bms and the slave bms communication via rs485! this can be seen in the photo) and so, at the moment when the inverter is working with the master bms + slave bms, we can still receive data on the PC via rs485

Ah okay, excellent. This is exactly what I was hoping it would do. Not interfere with the CANBUS but still allow me to talk with the BMS for HA.

Do you have to use on of the slave RS485, or the one next to the CANBUS. I've also seen people split the CANBUS....? Thanks

tlamoureux24, 2025-04-21:
Hello, I have exactly the same setup.
The Windows application switches to parallel mode in this case, and from what I've observed, it only listens to the RS485 bus; no settings on any BMS can be changed.
But it's still a useful mode for those who just want to monitor their BMSs.

[jk-bms-monitor_log_in_parallel.txt](https://github.com/user-attachments/files/19830396/jk-bms-monitor_log_in_parallel.txt)

Silverwings007, 2025-06-19:
Has anyone found a solution to reading the data off a slave and master at the same time? I don't need to change any settings, just read data.

Thanks

BelTrappist, 2025-07-19:
I’m also interested to read my batteries in HA. Similar setup: 2x PB2A16S20P, currently in Master and Slave. 
I have not been successful (Yet) to setup the communication to my inverter correctly.
I’m a real newbie, still trying to figure out how to read one battery…
I did succeed to program another esp32 to read my Inverter data into my HASS, but since the connection to the inverter is not successful, the battery reading is limited and wrong (SOC for example). So if you find the way to either read everything through the Master or by connecting both through UART protocol on the ESP32 it would be great.
I also wonder if I could use one ESP32 to get inverter and batteries data, simplifying the wiring, but probably getting more difficult to program (and beyong my capabilities for sure).

vpritulin, 2025-10-15:
@fastvd   Hi, did you manage to receive data from the master and slave in esphome?