syssi/esphome-jk-bms #707: New version Issue?

100000watt, 2025-03-22:
Hello, I am using your code to read some Neey v4 balancers. The devices created a year ago connect and work fine, but when trying to create another device in these days, it refuses to connect to the BT of the new Neey. I have tried more than one ESP module and attempted to connect to an old Neey, thinking that the protocol might have changed on the new one. Nothing works; Wi-Fi is fine, but Bluetooth always returns an error. Is it possible that this is a bug in the new version of the code?
Thank you

syssi, 2025-03-22:
> Is it possible that this is a bug in the new version of the code?

What do you mean exactly? IMO this project isn't broken. You could flash another ESP32 and try to talk to your old balancer to verify it's working in general. I'm not aware of any changes.

100000watt, 2025-03-22:
Thanks for the reply, yes, I’ve already tried 3 different ESPs with the Wroom-32 module, and I’ve also tried connecting to the older Neey. I can post the log here?

syssi, 2025-03-22:
Sure. Please post the logs. :-)

100000watt, 2025-03-22:
[logs_esp32-heltec-balancer-ble-example_logs.txt](https://github.com/user-attachments/files/19402322/logs_esp32-heltec-balancer-ble-example_logs.txt)

syssi, 2025-03-22:
Could you provide the MAC address of your old and the new balancer?

syssi, 2025-03-22:
This is the important part from the log:

```
```
[11:08:37][I][esp32_ble_client:110]: [0] [2B:80:03:F5:8C:B4] 0x00 Attempting BLE connection
[11:08:37][D][esp-idf:000][BTU_TASK]: W (79981) BT_HCI: hcif disc complete: hdl 0x0, rsn 0x3e
[11:08:37][D][esp-idf:000][BTU_TASK]: W (80083) BT_HCI: hcif disc complete: hdl 0x0, rsn 0x3e
[11:08:37][D][esp-idf:000][BTU_TASK]: W (80191) BT_HCI: hcif disc complete: hdl 0x0, rsn 0x3e
[11:08:37][D][esp-idf:000][BTU_TASK]: W (80295) BT_APPL: gattc_conn_cb: if=3 st=0 id=3 rsn=0x3e
[11:08:37][D][esp-idf:000][BTU_TASK]: W (80301) BT_HCI: hcif disc complete: hdl 0x0, rsn 0x3e
[11:08:37][E][esp32_ble_client:238]: [0] [2B:80:03:F5:8C:B4] Got ESP_GATTC_OPEN_EVT while not in connecting state, status=133
```

Reason `0x3e` typically refers to a connection timeout or device not in range.

100000watt, 2025-03-22:
The new Neey have 2B:80:03:F5:8C:B4 mac address, the old have 00:E4:4C:21:D9:69 mac. 
I also tried placing the ESP module just a few centimeters from the device, but the connection is still completely unstable. Once the connection is established, it drops within a few seconds.
I don't understand why, though, the ESP module created a year ago works, as they were purchased together at the time. The only thing that has changed is having re-downloaded the code from your GitHub

100000watt, 2025-03-22:
Sorry for the many questions, what does this warning mean?
INFO Connected to MQTT broker!
[12:02:56][I][mqtt:291]: MQTT Connected!
[12:03:09][W][component:170]: Component mqtt cleared Warning flag
[12:03:09][I][mqtt:291]: MQTT Connected!
[12:03:29][W][component:170]: Component mqtt cleared Warning flag
[12:03:29][I][mqtt:291]: MQTT Connected!

syssi, 2025-03-22:
> Sorry for the many questions, what does this warning mean?
> [12:03:09][W][component:170]: Component mqtt cleared Warning flag

I don't know. IMO it's unrelated.

syssi, 2025-03-22:
I've uplodated a fresh ESHPHome 2025.3.0 build to my ESP32 (monitoring a JK-BMS) using the same BLE client connection scheme. It does work out of the box.

syssi, 2025-03-22:
Please keep in mind for close is as bad as to far away. You have to find a sweet shot for the distance between the BMS and the ESP32.

100000watt, 2025-03-22:
Understood, I do some tests then at different distances. Thank you, very kind. 
p.s. if I ever wanted to make a serial version by removing the BT module on the balancer it would be really welcome. ❤

100000watt, 2025-03-22:
To run a test, is it possible to compile the code using an older version from a local repository? For example, version 1.5.0 that runs on my old ESP.
I'm trying everything, but now this problem has been added:

[13:41:01][W][heltec_balancer_ble:252]: Frame dropped because of invalid length
[13:41:01][W][heltec_balancer_ble:270]: CRC check failed! 0x10 != 0x6F
[13:41:03][W][heltec_balancer_ble:252]: Frame dropped because of invalid length
[13:41:04][W][heltec_balancer_ble:252]: Frame dropped because of invalid length
[13:41:04][W][heltec_balancer_ble:270]: CRC check failed! 0x1B != 0x74
[13:41:05][W][heltec_balancer_ble:270]: CRC check failed! 0x89 != 0xE4
[13:41:07][W][heltec_balancer_ble:252]: Frame dropped because of invalid length
[13:41:08][W][heltec_balancer_ble:252]: Frame dropped because of invalid length
[13:41:09][W][heltec_balancer_ble:252]: Frame dropped because of invalid length
[13:41:10][W][heltec_balancer_ble:252]: Frame dropped because of invalid length
[13:41:11][W][heltec_balancer_ble:270]: CRC check failed! 0xDE != 0xD7

syssi, 2025-03-22:
> To run a test, is it possible to compile the code using an older version from a local repository? For example, version 1.5.0 that runs on my old ESP.

Yes. Per default the `main` branch is used but you can pick a specific release here:

```
substitutions:
  external_components_source: github://syssi/esphome-jk-bms@main
```

vs.

```
substitutions:
  external_components_source: github://syssi/esphome-jk-bms@1.5.0
```

You can find the different release tags here: https://github.com/syssi/esphome-jk-bms/releases

But please be aware: Using a different component version is just going back to an old release of this project / implementation. I guess my implementation isn't the root causes because the `ble_client` connection fails already and my component is involved after the established conneciton.

syssi, 2025-03-22:
This is a new issue:

```
[13:41:01][W][heltec_balancer_ble:252]: Frame dropped because of invalid length
[13:41:01][W][heltec_balancer_ble:270]: CRC check failed! 0x10 != 0x6F
[13:41:03][W][heltec_balancer_ble:252]: Frame dropped because of invalid length
[13:41:04][W][heltec_balancer_ble:252]: Frame dropped because of invalid length
[13:41:04][W][heltec_balancer_ble:270]: CRC check failed! 0x1B != 0x74
[13:41:05][W][heltec_balancer_ble:270]: CRC check failed! 0x89 != 0xE4
[13:41:07][W][heltec_balancer_ble:252]: Frame dropped because of invalid length
[13:41:08][W][heltec_balancer_ble:252]: Frame dropped because of invalid length
[13:41:09][W][heltec_balancer_ble:252]: Frame dropped because of invalid length
[13:41:10][W][heltec_balancer_ble:252]: Frame dropped because of invalid length
[13:41:11][W][heltec_balancer_ble:270]: CRC check failed! 0xDE != 0xD7
```

It looks like the connection is established properly and this project isn't able to decode the balancer traffic. Could you increase the log level to `VERY_VERBOSE` and provide another complete ESPHome log? I would like to see the raw traffic which cannot be decoded.

100000watt, 2025-03-22:
I attach the log with verbose debugging, but I think it has never connected to the BT. 
Now I try to move the ESP module again, now it is a couple of meters from the balancer and there is no obstacle between them

[logs_esp32-heltec-balancer-ble-example_logs (1).txt](https://github.com/user-attachments/files/19404354/logs_esp32-heltec-balancer-ble-example_logs.1.txt)

syssi, 2025-03-22:
```
[17:47:44][W][heltec_balancer_ble:239]: [2B:80:03:F5:8C:B4] Not connected
...
[17:52:58][W][heltec_balancer_ble:239]: [2B:80:03:F5:8C:B4] Not connected
```

The ESP32 wasn't able to establish a connection.

syssi, 2025-03-22:
Can you lookup the ESPHome version you did use as you flashed the project version `1.5.0`?

100000watt, 2025-03-22:
I don't know what version it was because the working devices were created a year ago and in the meantime I had to format the home assistant installation

100000watt, 2025-03-22:
Update: Compiling your main version with these settings connects

`esphome:
  name: ${name}
  comment: ${device_description}
  min_version: 2025.2.0
  project:
    name: "syssi.esphome-jk-bms"
    version: 5.1.0`

 but the CRC error remains. I'm attaching the log

100000watt, 2025-03-22:
[logs_esp32-heltec-balancer-ble-example_logs (2).txt](https://github.com/user-attachments/files/19405536/logs_esp32-heltec-balancer-ble-example_logs.2.txt)

syssi, 2025-03-23:
Could you increas the log level to `VERY VERBOSE`? I would like to see the raw traffic.

syssi, 2025-03-23:
These settings are info values only

```
esphome: name: ${name}
 comment: ${device_description}
  min_version: 2025.2.0
  project:
    name: "syssi.esphome-jk-bms"
    version: 5.1.0
```

They are used for advertising but don't change the release version. Could you show your complete YAML? I would like to see there is any important difference.

100000watt, 2025-03-23:
`substitutions:
  name: heltec-balancer
  device_description: "Monitor and control a Heltec/NEEY 4A balancer via bluetooth"
  external_components_source: github://syssi/esphome-jk-bms@main
  mac_address: 2B:80:03:F5:8C:B4

esphome:
  name: ${name}
  comment: ${device_description}
  min_version: 2025.2.0
  project:
    name: "syssi.esphome-jk-bms"
    version: 5.1.0

esp32:
  board: esp32dev
  framework:
    type: esp-idf
    version: latest

external_components:
  - source: ${external_components_source}
    refresh: 0s

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

ota:
  platform: esphome

logger:
  level: VERY_VERBOSE
  logs:
    esp32_ble_tracker: INFO
    esp32_ble_client: INFO

# If you don't use Home Assistant please remove this `api` section and uncomment the `mqtt` component!
#api:

mqtt:
   broker: !secret mqtt_host
   port: !secret mqtt_port
   username: !secret mqtt_username
   password: !secret mqtt_password
   id: mqtt_client

esp32_ble_tracker:
  scan_parameters:
    active: false

ble_client:
  - mac_address: ${mac_address}
    id: client0

heltec_balancer_ble:
  - ble_client_id: client0
    throttle: 1s
    id: balancer0

sensor:
  - platform: heltec_balancer_ble
    heltec_balancer_ble_id: balancer0
    min_cell_voltage:
      name: "${name} min cell voltage"
    max_cell_voltage:
      name: "${name} max cell voltage"
    delta_cell_voltage:
      name: "${name} delta cell voltage"
    cell_voltage_1:
      name: "${name} cell voltage 1"
    cell_voltage_2:
      name: "${name} cell voltage 2"
    cell_voltage_3:
      name: "${name} cell voltage 3"
    cell_voltage_4:
      name: "${name} cell voltage 4"
    cell_voltage_5:
      name: "${name} cell voltage 5"
    cell_voltage_6:
      name: "${name} cell voltage 6"
    cell_voltage_7:
      name: "${name} cell voltage 7"
    cell_voltage_8:
      name: "${name} cell voltage 8"
    cell_voltage_9:
      name: "${name} cell voltage 9"
    cell_voltage_10:
      name: "${name} cell voltage 10"
    cell_voltage_11:
      name: "${name} cell voltage 11"
    cell_voltage_12:
      name: "${name} cell voltage 12"
    cell_voltage_13:
      name: "${name} cell voltage 13"
    cell_voltage_14:
      name: "${name} cell voltage 14"
    cell_voltage_15:
      name: "${name} cell voltage 15"
    cell_voltage_16:
      name: "${name} cell voltage 16"
    total_voltage:
      name: "${name} total voltage"
    temperature_sensor_1:
      name: "${name} temperature sensor 1"
    total_runtime:
      name: "${name} total runtime"
    balancing_current:
      name: "${name} balancing current"
    # Not implemented
    # errors_bitmask:
    #   name: "${name} errors bitmask"
    
text_sensor:
  - platform: heltec_balancer_ble
    heltec_balancer_ble_id: balancer0
    # Not implemented
    # errors:
    #   name: "${name} errors"
    operation_status:
      name: "${name} operation status"
    `

100000watt, 2025-03-23:
This is the complete yaml file

[esp32-heltec-balancer-ble-example.txt](https://github.com/user-attachments/files/19408161/esp32-heltec-balancer-ble-example.txt)

100000watt, 2025-03-23:
the log level is configured to VERY VERBOSE

syssi, 2025-03-23:
Oh. I've pushed a change to print the raw notifications to the log:

https://github.com/syssi/esphome-jk-bms/commit/1105707680452d424c261fff793a67ae00a42a18

Please compile and flash your ESP another time and provide a new log. It should contain the raw data this time.

100000watt, 2025-03-23:
[logs_esp32-heltec-balancer-ble-example_logs (3).txt](https://github.com/user-attachments/files/19410496/logs_esp32-heltec-balancer-ble-example_logs.3.txt)

syssi, 2025-03-23:
It looks like a perfect connection this time:

```
[18:31:01][VV][heltec_balancer_ble:227]: Notification received: 55.AA.11.01.01.00.64.00.45.4B.2D.32.34.53.34.45.42.00.00.00 (20)
[18:31:01][VV][heltec_balancer_ble:227]: Notification received: 00.00.00.00.48.57.2D.33.2E.32.2E.30.5A.48.2D.32.2E.30.2E.32 (20)
[18:31:01][VV][heltec_balancer_ble:227]: Notification received: 56.31.2E.32.2E.32.37.00.32.30.32.34.30.31.30.34.06.00.00.00 (20)
[18:31:01][VV][heltec_balancer_ble:227]: Notification received: DD.DE.CD.00.32.42.38.30.30.33.46.35.38.43.42.34.31.32.33.34 (20)
[18:31:01][VV][heltec_balancer_ble:227]: Notification received: 35.36.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.9C.FF (20)
[18:31:01][I][heltec_balancer_ble:698]: Device info frame (100 bytes):
[18:31:01][D][heltec_balancer_ble:699]:   55.AA.11.01.01.00.64.00.45.4B.2D.32.34.53.34.45.42.00.00.00.00.00.00.00.48.57.2D.33.2E.32.2E.30.5A.48.2D.32.2E.30.2E.32.56.31.2E.32.2E.32.37.00.32.30.32.34.30.31.30.34.06.00.00.00.DD.DE.CD.00.32.42.38.30.30.33.46.35.38.43.42.34.31.32.33.34.35.36.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.9C.FF (100)
[18:31:01][I][heltec_balancer_ble:716]:   Model: EK-24S4EB
[18:31:01][I][heltec_balancer_ble:718]:   Hardware version: HW-3.2.0
[18:31:01][I][heltec_balancer_ble:720]:   Software version: ZH-2.0.2
[18:31:01][I][heltec_balancer_ble:722]:   Protocol version: V1.2.27
[18:31:01][I][heltec_balancer_ble:724]:   Manufacturing date: 20240104
[18:31:01][I][heltec_balancer_ble:726]:   Power on count: 6
[18:31:01][I][heltec_balancer_ble:728]:   Total runtime: 156d 3h (13491933s)
[18:31:03][I][heltec_balancer_ble:247]: Request status notification
[18:31:03][D][heltec_balancer_ble:800]: Write register: AA.55.11.01.02.00.14.00.00.00.00.00.00.00.00.00.00.00.27.FF (20)
[18:31:03][VV][heltec_balancer_ble:227]: Notification received: 55.AA.11.01.02.00.2C.01.6F.4B.27.52.40.93.33.52.40.26.21.52 (20)
[18:31:03][VV][heltec_balancer_ble:227]: Notification received: 40.C7.19.52.40.67.12.52.40.C7.19.52.40.E3.2F.52.40.A2.13.52 (20)
[18:31:03][VV][heltec_balancer_ble:227]: Notification received: 40.2E.11.52.40.3C.1C.52.40.10.26.52.40.1E.31.52.40.7E.0D.52 (20)
[18:31:04][VV][heltec_balancer_ble:227]: Notification received: 40.9A.23.52.40.D5.24.52.40.F2.3A.52.40.00.00.00.00.00.00.00 (20)
[18:31:04][VV][heltec_balancer_ble:227]: Notification received: 00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00 (20)
[18:31:04][VV][heltec_balancer_ble:227]: Notification received: 00.00.00.00.00.A9.08.04.3E.D4.CA.05.3E.0A.1E.08.3E.3A.6B.07 (20)
[18:31:04][VV][heltec_balancer_ble:227]: Notification received: 3E.54.27.05.3E.75.5F.06.3E.A0.37.07.3E.70.94.02.3E.45.EE.00 (20)
[18:31:04][VV][heltec_balancer_ble:227]: Notification received: 3E.B2.92.02.3E.BC.CC.01.3E.64.06.01.3E.3E.AF.00.3E.87.7F.03 (20)
[18:31:04][VV][heltec_balancer_ble:227]: Notification received: 3E.74.D8.01.3E.4A.9A.FD.3D.00.00.00.00.00.00.00.00.00.00.00 (20)
[18:31:04][VV][heltec_balancer_ble:227]: Notification received: 00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00 (20)
[18:31:04][VV][heltec_balancer_ble:227]: Notification received: 00.AF.21.52.42.AF.21.52.40.00.D0.35.3B.0F.0C.0F.0A.00.00.00 (20)
[18:31:04][VV][heltec_balancer_ble:227]: Notification received: 00.B9.1E.97.41.B9.1E.97.41.00.00.00.00.00.00.00.00.00.00.00 (20)
[18:31:04][VV][heltec_balancer_ble:227]: Notification received: 00.00.00.00.00.00.00.00.00.00.00.00.00.00.70.8C.CC.00.00.00 (20)
[18:31:04][VV][heltec_balancer_ble:227]: Notification received: 00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00 (20)
[18:31:04][VV][heltec_balancer_ble:227]: Notification received: 00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.19.FF (20)
[18:31:04][I][heltec_balancer_ble:327]: Cell info frame (300 bytes):
[18:31:04][D][heltec_balancer_ble:328]:   55.AA.11.01.02.00.2C.01.6F.4B.27.52.40.93.33.52.40.26.21.52.40.C7.19.52.40.67.12.52.40.C7.19.52.40.E3.2F.52.40.A2.13.52.40.2E.11.52.40.3C.1C.52.40.10.26.52.40.1E.31.52.40.7E.0D.52.40.9A.23.52.40.D5.24.52.40.F2.3A.52.40.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.A9.08.04.3E.D4.CA.05.3E.0A.1E.08.3E.3A.6B.07.3E.54.27.05.3E.75.5F.06.3E.A0.37.07.3E.70.94.02.3E.45.EE.00.3E.B2.92.02.3E.BC.CC.01.3E.64 (150)
[18:31:04][D][heltec_balancer_ble:329]:   06.01.3E.3E.AF.00.3E.87.7F.03.3E.74.D8.01.3E.4A.9A.FD.3D.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.AF.21.52.42.AF.21.52.40.00.D0.35.3B.0F.0C.0F.0A.00.00.00.00.B9.1E.97.41.B9.1E.97.41.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.70.8C.CC.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.19.FF (150)
[18:31:04][D][heltec_balancer_ble:356]:   Frame counter: 111
[18:31:05][I][heltec_balancer_ble:514]:   Uptime: 155d 3h (13405296s)
[18:31:05][VV][heltec_balancer_ble:227]: Notification received: 55.AA.11.01.02.00.2C.01.70.BF.29.52.40.09.36.52.40.9A.23.52 (20)
[18:31:05][VV][heltec_balancer_ble:227]: Notification received: 40.3C.1C.52.40.7E.0D.52.40.8C.18.52.40.59.32.52.40.DD.14.52 (20)
[18:31:05][VV][heltec_balancer_ble:227]: Notification received: 40.F3.0F.52.40.A2.13.52.40.26.21.52.40.E3.2F.52.40.CE.09.52 (20)
[18:31:05][VV][heltec_balancer_ble:227]: Notification received: 40.2D.3C.52.40.93.33.52.40.1E.31.52.40.00.00.00.00.00.00.00 (20)
[... cut ...]
[18:31:09][D][heltec_balancer_ble:329]:   06.01.3E.3E.AF.00.3E.87.7F.03.3E.74.D8.01.3E.4A.9A.FD.3D.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.DA.1B.52.42.DA.1B.52.40.00.38.82.3B.0D.05.0F.0A.00.00.00.00.B9.1E.97.41.B9.1E.97.41.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.72.8C.CC.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.58.FF (150)
[18:31:09][D][heltec_balancer_ble:356]:   Frame counter: 115
```

There is no CRC error at the log. Could you try to catch another CRC error?

100000watt, 2025-03-23:
I noticed this too, I can't explain why sometimes it does it and other times it doesn't.
How sometimes it doesn't want to connect and other times it connects immediately at the same distance, device etc...
It's a mystery

syssi, 2025-03-23:
I try to explain some more details because we a looking at different issues on different levels at the moment:

1. If the ESPHome node is unable to establish a BLE connection to the Balancer only the `ble_client` component, the `esp32_ble` component and the `ESP-IDF` framework is involved. Downgrading ESPHome and/or changing the ESP-IDF framwork version can change the behaviour.
3. If the connection can be established the implementation of this project takes over the BLE client connection and sends instructions to the Balancer. The Balancer will respond with multiple BLE notifications per response. If a (permanent) CRC error occurs the situation can be improved probably by improving the implementation (of the frame assembly).

syssi, 2025-03-24:
Did you select the `latest` esp-idf version on purpose?

```
esp32:
  board: esp32dev
  framework:
    type: esp-idf
    version: latest
```

The default tag is called `recommended`.

[image]

100000watt, 2025-03-24:
I tried both

wetty, 2025-03-26:
Since a couple of days I also have some, maybe connection Problems. Everything is up to date. ESPHome Device Builder Version: 2025.3.2

My setup with two JK BMS is running since 1-2 years. No changes in hardware or location of my device
esp32:
  board: wemos_d1_mini32
  framework:
    type: esp-idf
    version: latest

It seems that ESPhome sets some values in homeassistant, which were not received from the bms. 
And my balance starting voltage was 3,43V until 4 weeks ago. I changed it several times in homeassistant to 3.4V and also 2-3 times via the JK BMS App directly on the BMS. homeassistant always returns to 3.43V and overrides the settings in the BMS.
Should I create a new Issue?
What kind of logs do you need?

[image]

[09:50:15][D][sensor:093]: 'jkbms bms1 cell resistance 7': Sending state 0.05500 Ω with 3 decimals of accuracy
[09:50:15][D][sensor:093]: 'jkbms bms1 cell voltage 8': Sending state 3.30900 V with 3 decimals of accuracy
[09:50:15][D][sensor:093]: 'jkbms bms1 cell resistance 8': Sending state 0.05500 Ω with 3 decimals of accuracy
[09:50:15][D][sensor:093]: 'jkbms bms1 errors bitmask': Sending state 0.00000  with 0 decimals of accuracy
[09:50:16][D][text_sensor:064]: 'jkbms bms1 errors': Sending state ''
[09:50:16][D][sensor:093]: 'jkbms bms1 balancing current': Sending state 0.00000 A with 2 decimals of accuracy
[09:50:16][D][jk_bms_ble:549]: Precharge: OFF
[09:50:16][D][jk_bms_ble:552]: User alarm: 0x00 0x00 (always 0xC5 0x09?)
[09:50:16][D][sensor:093]: 'jkbms bms1 total runtime': Sending state 49637004.00000 s with 0 decimals of accuracy
[09:50:16][D][text_sensor:064]: 'jkbms bms1 total runtime formatted': Sending state '1y 209d 12h'
[09:50:16][D][jk_bms_ble:569]: Balancing indicator (new): NO
[09:50:16][D][jk_bms_ble:595]: Heating current: 6.870 A
[09:50:16][D][jk_bms_ble:598]: Charger Plugged: OFF
[09:50:16][D][jk_bms_ble:599]: Temperature sensor 3: 23.0 °C
[09:50:16][D][jk_bms_ble:600]: Temperature sensor 4: 0.0 °C
[09:50:16][D][jk_bms_ble:601]: Temperature sensor 5: 0.0 °C
[09:50:16][D][sensor:093]: 'jkbms bms0 cell resistance 8': Sending state 0.07600 Ω with 3 decimals of accuracy
[09:50:16][D][sensor:093]: 'jkbms bms0 cell voltage 9': Sending state 3.30900 V with 3 decimals of accuracy
[09:50:16][D][sensor:093]: 'jkbms bms0 cell resistance 9': Sending state 0.07900 Ω with 3 decimals of accuracy
[09:50:16][D][sensor:093]: 'jkbms bms0 charging power': Sending state 549.69934 W with 2 decimals of accuracy
[09:50:16][D][sensor:093]: 'jkbms bms0 discharging power': Sending state 0.00000 W with 2 decimals of accuracy
[09:50:17][D][sensor:093]: 'jkbms bms0 temperature sensor 1': Sending state 20.10000 °C with 1 decimals of accuracy
[09:50:17][W][component:237]: Component esp32_ble took a long time for an operation (1737 ms).
[09:50:17][W][component:238]: Components should block for at most 30 ms.
[09:50:17][W][esp32_ble_tracker:125]: Too many BLE events to process. Some devices may not show up.
[09:50:17][C][jk_bms_ble.button:010]: JkBmsBle Button 'jkbms bms0 retrieve settings'
[09:50:17][C][jk_bms_ble.number:009]: JkBmsBle Number 'jkbms bms0 cell voltage overvoltage protection'
[09:50:17][C][jk_bms_ble.number:009]:   Unit of Measurement: 'V'
[09:50:17][C][jk_bms_ble.number:009]: JkBmsBle Number 'jkbms bms0 max discharge current'
[09:50:17][C][jk_bms_ble.number:009]:   Unit of Measurement: 'A'
[09:50:17][C][jk_bms_ble.number:009]: JkBmsBle Number 'jkbms bms1 cell voltage undervoltage protection'
[09:50:17][C][jk_bms_ble.number:009]: JkBmsBle Number 'jkbms bms1 total battery capacity'
[09:50:17][C][jk_bms_ble.number:009]:   Unit of Measurement: 'Ah'
[09:50:17][C][jk_bms_ble.number:009]: JkBmsBle Number 'jkbms bms1 balance starting voltage'
[09:50:17][C][jk_bms_ble.switch:068]: JkBmsBle Switch 'jkbms bms1 balancer'
[09:50:17][C][jk_bms_ble.switch:070]:   Icon: 'mdi:seesaw'
[09:50:17][C][jk_bms_ble.switch:090]:   Restore Mode: always OFF
[09:50:17][C][ble_switch:068]: BLE Client Switch 'jkbms bms0 enable bluetooth connection'
[09:50:17][C][ble_client:027]: BLE Client:
[09:50:17][C][mdns:116]: mDNS:
[09:50:17][C][mdns:117]:   Hostname: jkbms
[09:50:18][C][esphome.ota:073]: Over-The-Air updates:
[09:50:20][I][jk_bms_ble:344]: Cell info frame (version 3, 300 bytes) received
[09:50:20][D][sensor:093]: 'jkbms bms1 cell voltage 1': Sending state 3.30900 V with 3 decimals of accuracy
[09:50:20][D][sensor:093]: 'jkbms bms1 cell resistance 1': Sending state 0.05800 Ω with 3 decimals of accuracy
[09:50:20][D][sensor:093]: 'jkbms bms1 cell voltage 2': Sending state 3.31000 V with 3 decimals of accuracy
[09:50:20][D][sensor:093]: 'jkbms bms1 cell resistance 2': Sending state 0.05700 Ω with 3 decimals of accuracy

syssi, 2025-03-26:
@wetty Please create a new issue. This one is about the NEEY balancer + unstable/impossible BLE connection.

100000watt, 2025-03-26:
[image]

A new update was released today

wetty, 2025-03-26:
I installed the update before I reported this issue.

As I can get the most of the values via MQTT I may switch to that.

wetty, 2025-03-26:
> [@wetty](https://github.com/wetty) Please create a new issue. This one is about the NEEY balancer + unstable/impossible BLE connection.

Yes I will.

100000watt, 2025-03-26:
Small update, I tried to recompile today after updating esphome (yesterday I think). As if by magic it connects after a few seconds, the connection is stable, the only warning is a 

> [component:238]: Components should block for at most 30 ms.

For the rest it works perfectly

syssi, 2025-03-26:
The warning above can be ignored. This is logged for everybody using the project. It's because the decoding takes some time and isn't implemented async.

Could you summarize why it's working now and what was changed over time?

100000watt, 2025-03-26:
The only thing that has changed is that I installed today's esphome update, when I clicked install it re-downloaded a bunch of packages. But I can't say exactly what it downloaded

syssi, 2025-03-26:
Please monitor your setup some days and please reboot the ESP manually some times to make sure the connection can be always established.

100000watt, 2025-03-26:
Ok, I'll do some tests for a few days to understand if it's a case or not

Chris87-ux, 2025-04-04:
hi @syssi, @100000watt.
I also got some new Neey Balancer:

[image]
It seems, that I have the same issue.

But I run 6 Batterys and connected the 6 Neey to 2 Wroom32, the 1st has 2 oder versions of the neey which is running since years, the 2ns run one older one, which connect perfectly, but the 2 newer ( lik in the screenshot) dont connect. 

I saw, that my ESPhome is still on 2023.12.5.

What was the final solution to get the connection? --> only update ESPhome and compile again?

Thanks very much
/Chris

Chris87-ux, 2025-04-05:
Hi together, 

I updated now ESPhome andcompiled new. now i dont get any vales anymore, even not from the older Neey, see debug output:

[0;36m[D][esp32_ble_client:191]: [0] [00:E4:4C:21:DB:35] ESP_GATTC_NOTIFY_EVT[0m
05.04.2025 09:58:39 
[0;33m[W][heltec_balancer_ble:255]: Frame dropped because of invalid length[0m
05.04.2025 09:58:39(-0.04 seconds) 
[0;36m[D][esp32_ble_client:191]: [0] [00:E4:4C:21:DB:35] ESP_GATTC_NOTIFY_EVT[0m
05.04.2025 09:58:37(-2.57 seconds) 
[0;36m[D][esp32_ble_client:191]: [0] [00:E4:4C:21:DB:35] ESP_GATTC_NOTIFY_EVT[0m
05.04.2025 09:58:36(-0.91 seconds) 
[0;33m[W][heltec_balancer_ble:242]: [6E:A0:01:11:36:38] Not connected[0m
05.04.2025 09:58:36(-0.25 seconds) 
[0;33m[W][heltec_balancer_ble:242]: [1A:94:01:11:36:38] Not connected[0m
05.04.2025 09:58:36(-0.17 seconds) 
[0;33m[W][heltec_balancer_ble:255]: Frame dropped because of invalid length[0m
05.04.2025 09:58:36(-0 seconds) 
[0;36m[D][esp32_ble_client:191]: [0] [00:E4:4C:21:DB:35] ESP_GATTC_NOTIFY_EVT[0m
05.04.2025 09:58:33(-2.46 seconds) 
[0;36m[D][esp32_ble_client:191]: [0] [00:E4:4C:21:DB:35] ESP_GATTC_NOTIFY_EVT[0m
05.04.2025 09:58:31(-2.12 seconds) 
[0;33m[W][heltec_balancer_ble:242]: [6E:A0:01:11:36:38] Not connected[0m
05.04.2025 09:58:31(-0.2 seconds) 
[0;33m[W][heltec_balancer_ble:255]: Frame dropped because of invalid length[0m
05.04.2025 09:58:31(-0 seconds) 
[0;36m[D][esp32_ble_client:191]: [0] [00:E4:4C:21:DB:35] ESP_GATTC_NOTIFY_EVT[0m
05.04.2025 09:58:31(-0.11 seconds) 
[0;33m[W][heltec_balancer_ble:242]: [1A:94:01:11:36:38] Not connected[0m
05.04.2025 09:58:30(-0.75 seconds) 
[0;36m[D][esp32_ble_client:191]: [0] [00:E4:4C:21:DB:35] ESP_GATTC_NOTIFY_EVT[0m
05.04.2025 09:58:27(-3.35 seconds) 
[0;33m[W][heltec_balancer_ble:255]: Frame dropped because of invalid length[0m
05.04.2025 09:58:27(-0 seconds) 
[0;36m[D][esp32_ble_client:191]: [0] [00:E4:4C:21:DB:35] ESP_GATTC_NOTIFY_EVT[0m
05.04.2025 09:58:26(-0.56 seconds) 
[0;33m[W][heltec_balancer_ble:242]: [6E:A0:01:11:36:38] Not connected[0m
05.04.2025 09:58:26(-0.36 seconds) 
[0;33m[W][heltec_balancer_ble:242]: [1A:94:01:11:36:38] Not connected[0m
05.04.2025 09:58:25(-0.46 seconds) 
[0;36m[D][esp32_ble_client:191]: [0] [00:E4:4C:21:DB:35] ESP_GATTC_NOTIFY_EVT[0m

Thats my yaml:

[Neey_2.txt](https://github.com/user-attachments/files/19614423/Neey_2.txt)

Please help ;-)

/Chris

syssi, 2025-04-05:
Hmmm... it sounds like something (ESP-IDF version etc.) has changed under the hood which makes the situation worse. The NEEY component hasn't changed much recently. The root cause must be caused by the underlying libraries. Do you know how to downgrade ESPHome to play around with different versions/builds?

Chris87-ux, 2025-04-05:
Hi, unfortunately im not very deep into ESPhome. so i havent done any downgrades or so.

Regarding the 2 new Neeys, the status didint changed, there are still not able to connect.

syssi, 2025-06-29:
Please update to the most recent ESPHome version and give it another try!

syssi, 2025-07-04:
I close this issue because the initial question was resolved. If you have trouble / cannot establish a connection please feel free to create a new issue.

closed 2025-07-04 by syssi