syssi/esphome-jk-bms #713: Cannot connect to my BMS MAC

xtremeideaz, 2025-03-26:
Hello, Thanks for all you do. 

I have tried everything but it wont connect to my JK BMS MAC. 

[image]

[image]

xtremeideaz, 2025-04-07:
Hello @syssi please can you help add support for JKBMS ver Hardware ver V15A. The scanner doesnt find the ble MAC address. 

Please help.

Psynosaur, 2025-04-08:
Hi

Which yaml file did you use for flashing to ESP?

I used `esp32-ble-v14-example.yaml` on my `V15A` with software `V15.32` and it works just fine.

[image]

[image]

Cheers

xtremeideaz, 2025-04-10:
> Hi
> 
> Which yaml file did you use for flashing to ESP?
> 
> I used `esp32-ble-v14-example.yaml` on my `V15A` with software `V15.32` and it works just fine.
> 
> [image]
> 
> [image]
> 
> Cheers

I use ver 11, let me try this option and revert. thanks

syssi, 2025-05-27:
Did you resolve the issue?