syssi/esphome-jk-bms #719: ESPHome loosing entitys

MStapelfeldt, 2025-04-07:
Hi there,

I've been using the BLE version with HA for over a year now, and everything was working perfectly. Recently, however, I've encountered a strange issue: ESPHome keeps losing entities throughout the day.

When I check the dashboard, many sensors are listed as unavailable. Looking into the integration, I see that only 26–77 entities are available. However, when I reload the integration, the number jumps back to 119 entities, and everything works fine again. (Sometimes I need to reload the integration two or three times before all entities become visible again.) This fix lasts anywhere from 2 to 24 hours before I need to reload again.

Could someone help me debug this issue and find a way to restore stable functionality? I'm currently using `esp32-ble-v14-example.yaml` V2.1.0 with only minor adjustments, such as the MAC address and:
```
  platformio_options:
    build_flags: 
      - -DCONFIG_ARDUINO_LOOP_STACK_SIZE=32768
```

syssi, 2025-04-07:
Did you update the ESPHome version and flashed your ESPHome node recently? Do you use the native `api` or `mqtt`?

MStapelfeldt, 2025-04-07:
ESPHome and HA Versions are allways up to date, so it's ESPHome 2025.3.3 and Core 2025.4.0, Supervisor 2025.03.4 at the moment. I have the problem around a month now, so it's ESPHome 2025.2.x or .3.0 I first regonized it. I'm using native API. I also uncommented some Sensors, so just in case, here is the full YAML:
```yaml
substitutions:
  name: jk-bms
  device_description: "Monitor a JK-BMS via bluetooth"
  external_components_source: github://syssi/esphome-jk-bms@main
  mac_address: C8:47:80:03:DB:7F
  protocol_version: JK02_32S

esphome:
  name: ${name}
  comment: ${device_description}
  min_version: 2024.6.0
  project:
    name: "syssi.esphome-jk-bms"
    version: 2.1.0
  platformio_options:
    build_flags: 
      - -DCONFIG_ARDUINO_LOOP_STACK_SIZE=32768

esp32:
  board: wemos_d1_mini32
  framework:
    type: esp-idf

external_components:
  - source: ${external_components_source}
    refresh: 0s

wifi:
  networks:
    - ssid: !secret wifi_ssid
      password: !secret wifi_password

api:
  encryption:
    key: "g7EdOcBK6XYIAG00N/STCAt6bDODTgWXd4SjNSSsXqg="

ota:
  - platform: esphome
    password: !secret ota_passwd
    on_begin:
      then:
        - switch.turn_off: ble_client_switch0
        - logger.log: "BLE connection suspended for OTA update"

logger:
  level: DEBUG
  logs:
    esp32_ble_tracker: INFO
    esp32_ble_client: INFO

time:
  - platform: sntp

esp32_ble_tracker:
  scan_parameters:
    active: false

ble_client:
  - mac_address: ${mac_address}
    id: client0

jk_bms_ble:
  - ble_client_id: client0
    protocol_version: ${protocol_version}
    throttle: 5s
    id: bms0

binary_sensor:
[... 265 lines cut ...]
    #   name: "${name} disable temperature sensors"
    display_always_on:
      name: "${name} display always on"
    smart_sleep:
      name: "${name} smart sleep"
    disable_pcl_module:
      name: "${name} disable pcl module"
    timed_stored_data:
      name: "${name} timed stored data"
    charging_float_mode:
      name: "${name} charging float mode"

  - platform: ble_client
    ble_client_id: client0
    id: ble_client_switch0
    name: "${name} enable bluetooth connection"

text_sensor:
  - platform: jk_bms_ble
    errors:
      name: "${name} errors"
    total_runtime_formatted:
      name: "${name} total runtime formatted"
```
Really strange I think is: JK-BMS is the only device which is loosing entities and they are all back after reloading integration from ESPHome.

syssi, 2025-04-07:
Did you flash the YAML to a second ESP32 on your network by accident?

MStapelfeldt, 2025-04-07:
No, there is only one esp32 in my network connected, also this device is visible in HA, ESPHome and Dashboard. Then I see some entities unavaible, reload integration, all visible again. At the moment it's showing 103 entities, pressing reload: 108
Same device, not switching view or side or anything...

[image]
[image]

syssi, 2025-04-07:
Could you check your Home Assistant log for errors? If you remove the ESPHome node and re-add the device. Does it fix the issue?

MStapelfeldt, 2025-04-07:
Found these 2 Errors:
```
Logger: aioesphomeapi.connection
Quelle: runner.py:154
Erstmals aufgetreten: 4. April 2025 um 14:24:15 (224 Vorkommnisse)
Zuletzt protokolliert: 10:13:33
jk-bms @ 192.168.178.25: Connection error occurred: [Errno 104] Connection reset by peer
jk-bms @ 192.168.178.25: Connection error occurred: Ping response not received after 90.0 seconds
jk-bms @ 192.168.178.25: Connection error occurred: jk-bms @ 192.168.178.25: EOF received
shine_x_1 @ 192.168.178.26: Connection error occurred: [Errno 104] Connection reset by peer
jk-bms @ 192.168.178.25: Connection error occurred: jk-bms @ 192.168.178.25: Connection lost
```
```
Logger: aioesphomeapi.connection
Quelle: /usr/src/homeassistant/homeassistant/components/esphome/manager.py:874
Erstmals aufgetreten: 5. April 2025 um 12:56:13 (5 Vorkommnisse)
Zuletzt protokolliert: 12:48:57
jk-bms @ 192.168.178.25: disconnect request failed
Traceback (most recent call last):
  File "aioesphomeapi/connection.py", line 1026, in aioesphomeapi.connection.APIConnection.disconnect
  File "aioesphomeapi/connection.py", line 838, in send_message_await_response
  File "aioesphomeapi/connection.py", line 820, in aioesphomeapi.connection.APIConnection.send_messages_await_response_complex
aioesphomeapi.core.SocketClosedAPIError: jk-bms @ 192.168.178.25: EOF received
```
Not sure if they are related... ATM I still have 108 entities but I'll readd the device.
....
Done. Deleted from ESPHome, readded:
[image]

And after pressing reload integration:
[image]

Got one new Warning after readding:
```
Logger: homeassistant.helpers.script
Quelle: helpers/script.py:697
Erstmals aufgetreten: 3. April 2025 um 18:40:40 (60786 Vorkommnisse)
Zuletzt protokolliert: 14:14:35
Error in 'choose[0]' evaluation: In 'numeric_state': In 'numeric_state' condition: unknown entity sensor.jk_bms_min_cell_voltage
```

MStapelfeldt, 2025-04-10:
Hey, I've reduced the amount of entities to 88 to avoid memory issues and it was working for one day. Now I have same issue again, but I know which entries are lost:

[image]
[image]  

[image]
[image]

Any idea?

requies, 2025-05-21:
Same problem here since a couple of days.

syssi, 2025-05-21:
@requies Do you use the `api` or `mqtt` componen?

wetty, 2025-05-22:
Seems that with ESPHome Device Builder 2025.5.0 all the names changed from i.e.
jkbms_bms0_total_charging_cycle_capacity
jkbms_jkbms_bms0_total_charging_cycle_capacity
So an additionally jkbms was added

syssi, 2025-05-22:
Oh. May be the deprecation warning did break something here. I guess it will be fixed in `2025.5.1`.

syssi, 2025-06-29:
Please give it another try using `2025.6.2`. The native api was improved!