syssi/esphome-jk-bms #720: 3x UART ESP32  (02 BMS and 1 inverter) - jk_modbus Invalid header: 0x4E on specyfic circumstation

Kzibi, 2025-04-14:
2 BMS and Growatt SPH inverter - working well, no problems, except one circumstation.
When one batt ( name it Batt1) sending "Charging overvoltage" state, Batt2 is unavaiable immediately  with well known 'invalid header' errors.

Workaround - phisicaly disconnect Batt1 from inverter  and Batt2 is avaiable again and no 'invalid error' either.

Any idea ?

---
Z.K.

syssi, 2025-04-14:
I would try to isolate the connections to avoid common ground.

syssi, 2025-04-14:
A ADUM1201 per UART connection is recommended to avoid common ground issues.

syssi, 2025-04-15:
Keep me updated. Thanks in advance!

Kzibi, 2025-04-25:
So I got ADUM1201 modules. Now I wonder - how to supply power "+"  from BMS.
On connector pin is Vbat (50V) - it is very risky to step down 50V to 3V/5V. 
Offcourse I can get 3V form 1st cell (low voltage side) - it looks got:
1. 3.2V when normal MOSFET state
2.  3.5-4.5V when MOSFET is closed ( OverchargeVoltage state)

What do You think?

syssi, 2025-06-29:
The OEM RS485 converter dongle is using the VBAT pin. IMO this is the recommended power source.

syssi, 2025-07-04:
I close the issue because it must be resolved hardware-wise.

closed 2025-07-04 by syssi