syssi/esphome-jk-bms #727: JK-BMS PB Series to ESP32 via RS485-TTL: not working

doreks, 2025-04-27:
Hi everyone,

I wanted to share my setup for connecting a JK-BMS PB series (specifically the JK-PB2A16S20P model) to an ESP32 using RS485-TTL, along with the ESPHome configuration that worked for me. I hope this helps others looking to achieve a similar integration.

**🔌 Hardware Setup Components Used:**

JK-BMS PB series (JK-PB2A16S20P)

RS485 to TTL converter (3.3V logic level)

ESP32 development board

RJ45 cable with the following color coding:

Orange/White: A (RS485+)

Orange: B (RS485-)

Green/White: GND

**Wiring Diagram:**

`JK-BMS RJ45 Port       RS485 to TTL Converter       ESP32
----------------       ----------------------       -----
Pin 1 (A)   --------->  RS485 A (+)           --->  N/A
Pin 2 (B)   --------->  RS485 B (-)           --->  N/A
Pin 7 (GND) --------->  GND                   --->  GND
                       TXD                   --->  GPIO16 (RX)
                       RXD                   <---  GPIO17 (TX)
                       VCC                   --->  3.3V
`

**On the JK BMS** app I have configured UART1 Protocol No: 0

**Here's the ESPHome YAML configuration I used:**

```
substitutions:
  name: jk-bms
  device_description: "Monitor a JK-BMS (JK-PB series) via Modbus"
  external_components_source: github://syssi/esphome-jk-bms@main
  tx_pin: GPIO16
  rx_pin: GPIO17

esphome:
  name: ${name}
  comment: ${device_description}
  min_version: 2024.6.0
  project:
    name: "syssi.esphome-jk-bms"
    version: 1.5.0

#esp32:
#  board: wemos_d1_mini32
#  framework:
#    type: esp-idf

esp32:
  board: esp32dev
  framework:
    type: arduino

# Enable Home Assistant API
api:
  encryption:
    key: "secret+secret"

ota:
  - platform: esphome
    password: "secret"
[... 135 lines cut ...]
[02:12:42][D][uart_debug:114]: >>> 01:03:12:78:00:01:01:6B
[02:12:42][D][uart_debug:114]: >>> 01:03:12:78:00:01:01:6B
[02:12:43][D][modbus_controller:040]: Modbus command to device=1 register=0x1278 no response received - removed from send queue
[02:12:43][D][uart_debug:114]: >>> 01:03:12:7A:00:01:A0:AB
[02:12:43][D][uart_debug:114]: >>> 01:03:12:7A:00:01:A0:AB
[02:12:43][D][uart_debug:114]: >>> 01:03:12:7A:00:01:A0:AB
[02:12:43][D][uart_debug:114]: >>> 01:03:12:7A:00:01:A0:AB
[02:12:44][D][uart_debug:114]: >>> 01:03:12:7A:00:01:A0:AB
[02:12:44][D][modbus_controller:040]: Modbus command to device=1 register=0x127A no response received - removed from send queue
[02:12:44][D][uart_debug:114]: >>> 01:03:12:7C:00:01:40:AA
[02:12:44][W][modbus_controller:186]: Duplicate modbus command found: type=0x0 address=11233 count=1
[02:12:44][W][modbus_controller:186]: Duplicate modbus command found: type=0x0 address=37691 count=1
[02:12:44][W][modbus_controller:186]: Duplicate modbus command found: type=0x0 address=52724 count=1
[02:12:44][W][modbus_controller:186]: Duplicate modbus command found: type=0x3 address=4100 count=118
[02:12:44][W][modbus_controller:186]: Duplicate modbus command found: type=0x3 address=4108 count=2
[02:12:44][W][modbus_controller:186]: Duplicate modbus command found: type=0x3 address=4140 count=2
[02:12:44][W][modbus_controller:186]: Duplicate modbus command found: type=0x3 address=4152 count=2
[02:12:44][W][modbus_controller:186]: Duplicate modbus command found: type=0x3 address=4172 count=2
[02:12:44][W][modbus_controller:186]: Duplicate modbus command found: type=0x3 address=4180 count=2
[02:12:44][W][modbus_controller:186]: Duplicate modbus command found: type=0x3 address=4188 count=2
[02:12:44][W][modbus_controller:186]: Duplicate modbus command found: type=0x3 address=4196 count=2
[02:12:44][W][modbus_controller:186]: Duplicate modbus command found: type=0x3 address=4204 count=2
[02:12:44][W][modbus_controller:186]: Duplicate modbus command found: type=0x3 address=4608 count=32`

thanks for your time and support here!!

LithiumNL, 2025-06-23:
@doreks i have a qestion regarding your setup. How many bms's do you have? If you have multiple how are they connected to each other?
I'm trying to get a working example with 2 bms's, one in master and the other one in slave.

syssi, 2025-06-29:
@LithiumNL See https://github.com/syssi/esphome-jk-bms/discussions/753

You need a RS485 converter per BMS attached to the same ESP (on different `uart`s). The RS485 converter must be attached the the left RS485 port.

syssi, 2025-07-04:
@doreks Do you still need support?

ircoopr, 2025-07-12:
@doreks which BMS version did you try this on?

Are you certain that 'On the JK BMS app I have configured UART1 Protocol No: 0' worked? I thoght this was all setup for Protocol No:1?

closed 2025-07-25 by syssi