syssi/esphome-jk-bms #731: JK-BMS PB  ESP32 via RS485/Modbus  Doku?

Asathor85, 2025-05-04:
Hallo syssi,

leider scheitere ich daran deine esp32-jk-pb-modbus-example.yaml zum laufen zu bekommen.

Ich habe deine YAML unverändert bis auf den Webserver gelassen, aber es werden nur Fehlermeldungen zu Modbus Duplikate Commands und Modbus CRC Errors geworfen.

Die Link LED des BMS blinkt, der RS 485 empfängt auch permanent Daten.

Könntest du für weniger versierte Menschen eine kleine Doku machen, wie verkabelt wird ?
 Ich verstehe auch nicht wofür " flow_control_pin: GPIO12" gut sein soll....

Wäre dir sehr dankbar für deine Unterstützung.

syssi, 2025-05-04:
Hi! Mach mal ein Foto von deinem Setup insb. dem RS485 Converter und den verwendeten Ports am BMS und prüf das eingestellte Protokoll in der Android App. Danach versuche ich dir die einzelnen Details an deinem Setup zu erläutern.

Asathor85, 2025-05-04:
Danke für die Schnelle Rückmeldung.
Ports habe ich schon alle durch und auch mehrfach RX und TX getauscht vor lauter Ratlosigkeit.

[image]
[image]
[image]

syssi, 2025-05-04:
Der linke Port (RS485-1) entspricht `UART1` in der Android App. Hier ist bei dir aktuell das Protokoll `0` (000 - 4G-GPS Remote module Common protocol V4.2) ausgewählt. Dieses Protokoll ist hier im Projekt über die Komponente `jk_bms` implementiert. Wir wollen aber echtes Modbus, deshalb stellen wir das Protokoll für UART1 auf `1` (001 - JK BMS RS485 Modbus V1.0).

Anschließend spricht das BMS ein Protokoll, welches mit der Implementierung `modbus_controller` aus dem ESPHome Kern kompatibel ist. Deine geflashte YAML-Konfiguration (`esp32-jk-pb-modbus-example.yaml`) ist hier die richtige Wahl.

Es gibt RS485 Konverter Module, welche wie ein Sprechfunkgerät funktionieren. Über eine "Taste" muss hier zwischen Senden und Empfangen umgeschaltet werden. Dies nennt sich Flusskontrolle (flow_control). Wenn man ein solches Modul besitzt, dann zieht man ein weiteres Kabel von einem freien GPIO des ESPs zum Modul und konfiguriert es entsprechend in der YAML-Konfiguration. Daraufhin kümmert sich die Implementierung darum die Sprechtaste zu drücken, wenn es was zu sagen gibt. In deinem Fall braucht du hier keine Rücksicht nehmen. Dein Modul macht die Umschaltung automatisch.

Als Nächstes wäre ein ESPHome Logbuch interessant, wenn es nach der Umstellung des Protokolls im BMS nicht unmittelbar klappt.

syssi, 2025-05-04:
Hier findest du ein tolles Projekt mit vielen Bildern und Dokumentation, welche sich auf das von dir eingesetzte Setup & Protokoll konzentriert: https://github.com/phinix-org/Multiple-JK-BMS-by-Modbus-RS485

Hier findest du auch viele weitere Register / ESPHome-Konfigurationsschnippsel um noch viel mehr Details aus deinem BMS herauszuholen und/oder steuerbar zu machen.

Asathor85, 2025-05-05:
Ich habe jetzt die esp32-example.yaml aufgespielt und das Protokoll wieder auf 0/GPS/TTL zurück gestellt. 
Dann funktioniert es mit der unveränderten Hardware optimal.
Es muss also irgendetwas im Code das Problem sein.

syssi, 2025-06-29:
Ich mache das Thema an dieser Stelle zu.

closed 2025-06-29 by syssi