syssi/esphome-jk-bms #736: Deprecated text_sensor.TEXT_SENSOR_SCHEMA, switch.SWITCH_SCHEMA, button.BUTTON_SCHEMA, number.NUMBER_SCHEMA

github-k8n, 2025-05-21:
The following warning are issued with esphome 2025.5.0

```
WARNING Using `text_sensor.TEXT_SENSOR_SCHEMA` is deprecated and will be removed in ESPHome 2025.11.0. Please use `text_sensor.text_sensor_schema(...)` instead. If you are seeing this, report an issue to the external_component author and ask them to update it. https://developers.esphome.io/blog/2025/05/14/_schema-deprecations/. Component using this schema: text_sensor
WARNING Using `text_sensor.TEXT_SENSOR_SCHEMA` is deprecated and will be removed in ESPHome 2025.11.0. Please use `text_sensor.text_sensor_schema(...)` instead. If you are seeing this, report an issue to the external_component author and ask them to update it. https://developers.esphome.io/blog/2025/05/14/_schema-deprecations/. Component using this schema: text_sensor
WARNING Using `text_sensor.TEXT_SENSOR_SCHEMA` is deprecated and will be removed in ESPHome 2025.11.0. Please use `text_sensor.text_sensor_schema(...)` instead. If you are seeing this, report an issue to the external_component author and ask them to update it. https://developers.esphome.io/blog/2025/05/14/_schema-deprecations/. Component using this schema: text_sensor
WARNING Using `text_sensor.TEXT_SENSOR_SCHEMA` is deprecated and will be removed in ESPHome 2025.11.0. Please use `text_sensor.text_sensor_schema(...)` instead. If you are seeing this, report an issue to the external_component author and ask them to update it. https://developers.esphome.io/blog/2025/05/14/_schema-deprecations/. Component using this schema: text_sensor
WARNING Using `switch.SWITCH_SCHEMA` is deprecated and will be removed in ESPHome 2025.11.0. Please use `switch.switch_schema(...)` instead. If you are seeing this, report an issue to the external_component author and ask them to update it. https://developers.esphome.io/blog/2025/05/14/_schema-deprecations/. Component using this schema: heltec_balancer_ble
WARNING Using `button.BUTTON_SCHEMA` is deprecated and will be removed in ESPHome 2025.11.0. Please use `button.button_schema(...)` instead. If you are seeing this, report an issue to the external_component author and ask them to update it. https://developers.esphome.io/blog/2025/05/14/_schema-deprecations/. Component using this schema: heltec_balancer_ble
WARNING Using `button.BUTTON_SCHEMA` is deprecated and will be removed in ESPHome 2025.11.0. Please use `button.button_schema(...)` instead. If you are seeing this, report an issue to the external_component author and ask them to update it. https://developers.esphome.io/blog/2025/05/14/_schema-deprecations/. Component using this schema: heltec_balancer_ble
WARNING Using `button.BUTTON_SCHEMA` is deprecated and will be removed in ESPHome 2025.11.0. Please use `button.button_schema(...)` instead. If you are seeing this, report an issue to the external_component author and ask them to update it. https://developers.esphome.io/blog/2025/05/14/_schema-deprecations/. Component using this schema: heltec_balancer_ble
WARNING Using `number.NUMBER_SCHEMA` is deprecated and will be removed in ESPHome 2025.11.0. Please use `number.number_schema(...)` instead. If you are seeing this, report an issue to the external_component author and ask them to update it. https://developers.esphome.io/blog/2025/05/14/_schema-deprecations/. Component using this schema: heltec_balancer_ble
WARNING Using `number.NUMBER_SCHEMA` is deprecated and will be removed in ESPHome 2025.11.0. Please use `number.number_schema(...)` instead. If you are seeing this, report an issue to the external_component author and ask them to update it. https://developers.esphome.io/blog/2025/05/14/_schema-deprecations/. Component using this schema: heltec_balancer_ble
WARNING Using `number.NUMBER_SCHEMA` is deprecated and will be removed in ESPHome 2025.11.0. Please use `number.number_schema(...)` instead. If you are seeing this, report an issue to the external_component author and ask them to update it. https://developers.esphome.io/blog/2025/05/14/_schema-deprecations/. Component using this schema: heltec_balancer_ble
WARNING Using `number.NUMBER_SCHEMA` is deprecated and will be removed in ESPHome 2025.11.0. Please use `number.number_schema(...)` instead. If you are seeing this, report an issue to the external_component author and ask them to update it. https://developers.esphome.io/blog/2025/05/14/_schema-deprecations/. Component using this schema: heltec_balancer_ble
WARNING Using `number.NUMBER_SCHEMA` is deprecated and will be removed in ESPHome 2025.11.0. Please use `number.number_schema(...)` instead. If you are seeing this, report an issue to the external_component author and ask them to update it. https://developers.esphome.io/blog/2025/05/14/_schema-deprecations/. Component using this schema: heltec_balancer_ble
WARNING Using `number.NUMBER_SCHEMA` is deprecated and will be removed in ESPHome 2025.11.0. Please use `number.number_schema(...)` instead. If you are seeing this, report an issue to the external_component author and ask them to update it. https://developers.esphome.io/blog/2025/05/14/_schema-deprecations/. Component using this schema: heltec_balancer_ble
```

AnKloss, 2025-05-23:
Hello
My to,  so I restore 2025.4.2 back and is run normal as before. But I think is not a long time solution.
Hope to get soon a fix update.
mfg 
Andreas

syssi, 2025-05-23:
Some users are complaining about unavailable entities since ESPHome 2025.5.0. @AnKloss could you double check the device tile. There are also users reporting the entity names has changed which also causes confusion.

@github-k8n Does your setup still work including all entities? Maybe the new warning introduced a bug.

github-k8n, 2025-05-23:
@syssi yes, all my configured entities still work, no issue at all.
I typically do a "Clean Build Files" before updating .. maybe that's why?