syssi/esphome-jk-bms #755: No decimal point in the values after update Home Assistant Core to 2025.6.1

ans-gw, 2025-06-21:
No decimal point in the values,issue from homeaasistant update. 2025.6.1 !
Has been intruduced  suggested_display_precision
[See this article : Sensor device classes now have default display precision](https://developers.home-assistant.io/blog/2025/05/26/sensor-default-display-precision/).

Now all cell valie in Ha are 3 V.

Can you please implement this in the jk bms code,or give a suggestion on how to fix this problem?

Thank you very much

syssi, 2025-06-21:
This is a regression because of the new feature. The `esphome` integration of Home Assistant doesn't support the new behavior yet. I guess this project/component doesn't need to be changed. Let's wait some days/weeks.

ans-gw, 2025-06-21:
Hi,thanks for quick reply.
Yes I know is Ha problem for the new Ha ui setting.
https://developers.home-assistant.io/blog/2025/05/26/sensor-default-display-precision/ .

But i didnt know esphome does not support the new  Ha default-display-precision?!...this has not sense..
Could be very dangerous for lithium batteries !

P.S::
since Esphome last update ,(they change something about Ble) I have many of this lines in Esp Home Builder Log:
[14:32:40][D][esp32_ble_client:193]: [0] [Mac Adrress] ESP_GATTC_NOTIFY_EVT
[14:32:40][D][esp32_ble_client:193]: [0] [Mac Address] ESP_GATTC_NOTIFY_EVT

are this errors?,,can cause problems?

Thank you

syssi, 2025-06-21:
We are talking about "formatting at the user interface" here only. You can change it manually. Just the good default path doesn't work anymore:

[image]

ans-gw, 2025-06-21:
I try already.Whatever value I set there, exmple 3,32...it will keep show 3.32...not the real cell voltage!!.

what about [14:32:40][D][esp32_ble_client:193]: [0] [Mac Adrress] ESP_GATTC_NOTIFY_EVT ?

thank you

syssi, 2025-06-21:
```
[14:32:40][D][esp32_ble_client:193]: [0] [Mac Address] ESP_GATTC_NOTIFY_EVT
           ^Debug level
```

This is for debugging only. It indicates a incoming notification frame (with new status data) was received. Can be ignored!

syssi, 2025-06-21:
Why do you expect your setup doesn't work anymore?

ans-gw, 2025-06-21:
I remember before no have that lines shows many time,so i was wondering if that is a problem or not!

syssi, 2025-06-22:
In the past yes. At the moment this config isn't respected.

syssi, 2025-06-22:
This issue must be fixed at the ESPHome integration of HA core in my opinion. See https://github.com/home-assistant/core/issues/146873

AnKloss, 2025-06-27:
okay. Let wait HA fix it. I hope son as can. I use on my work HA now 5.3, no update at all) and on test Ha 6.3. Update just now. but is not fix .

Andreas

Macrisu, 2025-06-29:
HA does not intent to "fix" this.
It is the intended behaviour...

Macrisu, 2025-06-29:
Well, it seems so, as this is kind of important and I've just upgrade ESPHome Builder to 2025.6.2 and... no luck.

There are a lot of subjects on this issue, but no real answears for a quick fix.

AnKloss, 2025-06-29:
I go all ready back to home assistant 2025.5.3 and all working. Sins
 2025.6.0 this Desaster happen.

Best regards

Andreas

syssi, 2025-06-29:
This is the quick fix:

1. Go to the ESPHome device: Settings -> Devices -> a specific ESPHome Node
2. Scroll to an affected sensor. Mostly voltage sensors are affected
3. Click on a `cell voltage` 
4. Press the `gear` at the top left corner
5. Change the display precision from `3` to `3.333`

Repeat this for any cell voltage.

Macrisu, 2025-06-29:
I did that, but for 6 packs is frustrating!
6x16 minimum

Is there any way to modify in database/search&replace?

syssi, 2025-06-29:
I cannot help here. If I invest time in researching a simpler solution that saves you time, then you have simply outsourced the effort to me.

br3ily, 2025-06-29:
Thx, solved

> This is the quick fix:
>
> 1. Go to the ESPHome device: Settings -> Devices -> a specific ESPHome Node
> 2. Scroll to an affected sensor. Mostly voltage sensors are affected
> 3. Click on a `cell voltage` [...]

syssi, 2025-07-03:
This is the pull request to resolve the issue: https://github.com/home-assistant/core/pull/147849/files

It's part of HA core 2025.7.

closed 2025-07-03 by syssi

JohnGalt1717, 2025-07-03:
[image]