syssi/esphome-jk-bms #756: V19 Not mapping Editable Settings

JohnGalt1717, 2025-06-21:
Here's my config:

```
substitutions:
  device_name: bmshost11
  battery1_name: "Battery 11"
  battery1_id: battery11
  name: "BMS 11"
  bms1_id: bmshost11
  mac_address: "XX:XX:XX:XX:XX:XX"  
  protocol_version: JK02_32S
  external_components_source: github://syssi/esphome-jk-bms@main

esphome:
  name: bmshost11
  friendly_name: BMS Host 11

esp32:
  board: esp32dev
  framework:
    type: esp-idf    
    sdkconfig_options:
      CONFIG_COMPILER_OPTIMIZATION_SIZE: y
      CONFIG_BT_BLE_42_FEATURES_SUPPORTED: y
      CONFIG_BT_BLE_50_FEATURES_SUPPORTED: y
      CONFIG_ESP_TASK_WDT_TIMEOUT_S: "20" 

external_components:
  - source: ${external_components_source}
    refresh: 0s

api:

ota:
  platform: esphome
  password: !secret esphome_ota_password

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  use_address: ${device_name}.XXX.YYY

logger:

web_server:
  port: 80

preferences:
  flash_write_interval: 0.5h

time:
  - platform: homeassistant
    id: homeassistant_time

esp32_ble_tracker:
  scan_parameters:
    interval: 500ms
    window: 150ms    
  # on_ble_advertise:
  #   then:
  #     - lambda: |-
  #           ESP_LOGI("ble_adv", "  Name: %s", x.get_name().c_str());
  #           ESP_LOGI("ble_adv", "  MAC address: %s", x.address_str().c_str());    

 

ble_client:
  - mac_address: ${mac_address}
    id: ${bms1_id}

jk_bms_ble:
  - ble_client_id: ${bms1_id}
    throttle: 2s
    id: ${battery1_id}
    protocol_version: ${protocol_version}

binary_sensor:
  - platform: jk_bms_ble
    jk_bms_ble_id: ${battery1_id}
    balancing:
      name: "${name} balancing"
      on_press:
        then:
          - switch.turn_on: BMS7EXTBAL
      on_release:
        then:
          - switch.turn_off: BMS7EXTBAL
    charging:
      name: "${name} charging"
    discharging:
      name: "${name} discharging"
    online_status:
      name: "${name} online status"

button:
  - platform: jk_bms_ble
    jk_bms_ble_id: ${battery1_id}
    retrieve_settings:
      name: "${name} retrieve settings"
    retrieve_device_info:
      name: "${name} retrieve device info"
  - platform: template
    name: "Force Reconnect ${name}"
    on_press: 
      - lambda: |-
          id(${bms1_id})->connect();

  - platform: restart
    name: "Restart ${name} ESP32"
  - platform: shutdown
    name: "Shutdown ${name} ESP32"

number:
  - platform: jk_bms_ble
    jk_bms_ble_id: ${battery1_id}
    balance_trigger_voltage:
      name: "${name} balance trigger voltage"
    cell_count:
      name: "${name} cell count"
    total_battery_capacity:
      name: "${name} total battery capacity"
    cell_voltage_overvoltage_protection:
      name: "${name} cell voltage overvoltage protection"
    cell_voltage_overvoltage_recovery:
      name: "${name} cell voltage overvoltage recovery"
    cell_voltage_undervoltage_protection:
      name: "${name} cell voltage undervoltage protection"
    cell_voltage_undervoltage_recovery:
      name: "${name} cell voltage undervoltage recovery"
    balance_starting_voltage:
      name: "${name} balance starting voltage"
    voltage_calibration:
      name: "${name} voltage calibration"
    current_calibration:
      name: "${name} current calibration"
    power_off_voltage:
      name: "${name} power off voltage"
    max_balance_current:
      name: "${name} max balance current"
    max_charge_current:
      name: "${name} max charge current"
    max_discharge_current:
      name: "${name} max discharge current"

sensor:
  - platform: wifi_signal
    name: "WiFi Signal Sensor"
    update_interval: 60s

  - platform: total_daily_energy
    name: "${name} Total Charging Energy Today"
    power_id: bms_charging_power
    unit_of_measurement: kWh
    device_class: energy
    icon: mdi:counter
    accuracy_decimals: 3
    restore: true
    filters:
      - multiply: 0.001    
  - platform: total_daily_energy
    name: "${name} Total Discharging Energy Today"
    power_id: bms_discharging_power
    unit_of_measurement: kWh
    device_class: energy
    icon: mdi:counter
    accuracy_decimals: 3
    restore: true
    filters:
      - multiply: 0.001 

  - platform: jk_bms_ble
    jk_bms_ble_id: ${battery1_id}
    min_cell_voltage:
      name: "${name} min cell voltage"
    max_cell_voltage:
      name: "${name} max cell voltage"
    min_voltage_cell:
      name: "${name} min voltage cell"
    max_voltage_cell:
      name: "${name} max voltage cell"
    delta_cell_voltage:
      name: "${name} delta cell voltage"
    average_cell_voltage:
      name: "${name} average cell voltage"
    cell_voltage_1:
      name: "${name} cell voltage 1"
    cell_voltage_2:
      name: "${name} cell voltage 2"
    cell_voltage_3:
      name: "${name} cell voltage 3"
    cell_voltage_4:
      name: "${name} cell voltage 4"
    cell_voltage_5:
      name: "${name} cell voltage 5"
    cell_voltage_6:
      name: "${name} cell voltage 6"
    cell_voltage_7:
      name: "${name} cell voltage 7"
    cell_voltage_8:
      name: "${name} cell voltage 8"
    cell_voltage_9:
      name: "${name} cell voltage 9"
    cell_voltage_10:
      name: "${name} cell voltage 10"
    cell_voltage_11:
      name: "${name} cell voltage 11"
    cell_voltage_12:
      name: "${name} cell voltage 12"
    cell_voltage_13:
      name: "${name} cell voltage 13"
    cell_voltage_14:
      name: "${name} cell voltage 14"
    cell_voltage_15:
      name: "${name} cell voltage 15"
    cell_voltage_16:
      name: "${name} cell voltage 16"
    cell_resistance_1:
      name: "${name} cell resistance 1"
    cell_resistance_2:
      name: "${name} cell resistance 2"
    cell_resistance_3:
      name: "${name} cell resistance 3"
    cell_resistance_4:
      name: "${name} cell resistance 4"
    cell_resistance_5:
      name: "${name} cell resistance 5"
    cell_resistance_6:
      name: "${name} cell resistance 6"
    cell_resistance_7:
      name: "${name} cell resistance 7"
    cell_resistance_8:
      name: "${name} cell resistance 8"
    cell_resistance_9:
      name: "${name} cell resistance 9"
    cell_resistance_10:
      name: "${name} cell resistance 10"
    cell_resistance_11:
      name: "${name} cell resistance 11"
    cell_resistance_12:
      name: "${name} cell resistance 12"
    cell_resistance_13:
      name: "${name} cell resistance 13"
    cell_resistance_14:
      name: "${name} cell resistance 14"
    cell_resistance_15:
      name: "${name} cell resistance 15"
    cell_resistance_16:
      name: "${name} cell resistance 16"
    total_voltage:
      name: "${name} total voltage"
    current:
      name: "${name} current"
    power:
      name: "${name} power"
    charging_power:
      name: "${name} charging power"
      id: bms_charging_power
    discharging_power:
      name: "${name} discharging power"
      id: bms_discharging_power
    temperature_sensor_1:
      name: "${name} temperature sensor 1"
    temperature_sensor_2:
      name: "${name} temperature sensor 2"
    power_tube_temperature:
      name: "${name} power tube temperature"
    state_of_charge:
      name: "${name} state of charge"
    capacity_remaining:
      name: "${name} capacity remaining"
    total_battery_capacity_setting:
      name: "${name} total battery capacity setting"
    charging_cycles:
      name: "${name} charging cycles"
    total_charging_cycle_capacity:
      name: "${name} total charging cycle capacity"
    total_runtime:
      name: "${name} total runtime"
    balancing_current:
      name: "${name} balancing current"
    errors_bitmask:
      name: "${name} errors bitmask"

switch:
  - platform: jk_bms_ble
    jk_bms_ble_id: ${battery1_id}
    charging:
      name: "${name} charge"
    discharging:
      name: "${name} discharge"
    balancer:
      name: "${name} balance"
  - platform: gpio
    id: BMS7EXTBAL
    name: "BMS 7 External Balancer"        
    pin: GPIO14
    inverted: True

text_sensor:
  - platform: jk_bms_ble
    jk_bms_ble_id: ${battery1_id}
    errors:
      name: "${name} errors"
    total_runtime_formatted:
      name: "${name} total runtime formatted"
  # - platform: ble_scanner
  #   name: "BLE Devices Scanner"

```

All of the readable settings come up but not the editable stuff (like balance start voltage etc.

The rest of my units are all v14, v15 or the old style.

JohnGalt1717, 2025-06-21:
Update: Cell voltage is always 3V too.

syssi, 2025-06-21:
Please take a look at #755 for the decimal place issue. It's a regression because of a new HA core feature.

syssi, 2025-06-22:
Yes. It looks like there is no (big) difference between the v14/v15 and v19 revision BLE protocol-vise. The mentioned issues should be similar for every known model.

closed 2025-06-25 by syssi