syssi/esphome-jk-bms #758: JK B2A25S-RP

christo1331, 2025-06-24:
first thanks to your incredible job!

I have a JK B2A25S bms witch is 25 cells BMS.

I use JK02_32S config meaning we can have 32cells reading no?

everything is working fine in a multiple  height bms reading 

the only issue i have is with the 25th cell :

esphome don't permit to write over 24 cells like this:

    cell_voltage_25:
      name: "${bms8} cell voltage 200"  

Is it normal, thanks

syssi, 2025-06-24:
How many cells does your device support officially? Could you provide an esphome log with `VERY_VERBOSE` logging enabled? I would like to extract some example traffic from the log to add the missing cells. At the moment only 24 cells are implemented

christo1331, 2025-06-24:
I'm not sure that's what you want , i send you also the entire log file, thanks.

settings frame:

[21:15:18][D][button:010]: 'jk-bms bms8 retrieve settings' Pressed.
[21:15:18][D][jk_bms_ble:1493]: Write register: AA.55.90.EB.96.00.00.00.00.00.00.00.00.00.00.00.00.00.00.10 (20)

device info frame:

[21:18:48][I][jk_bms_ble:230]: Request device info
[21:18:48][D][jk_bms_ble:1493]: Write register: AA.55.90.EB.97.00.00.00.00.00.00.00.00.00.00.00.00.00.00.11 (20)

[logs_esphome-web-28b574_logs (2).txt](https://github.com/user-attachments/files/20890229/logs_esphome-web-28b574_logs.2.txt)
[logs_esphome-web-28b574_logs (1).txt](https://github.com/user-attachments/files/20890230/logs_esphome-web-28b574_logs.1.txt)

syssi, 2025-06-24:
I've implemented cell voltage and resistance 25 upto 32: https://github.com/syssi/esphome-jk-bms/pull/759/files

Please give it a try by extending your configuration. Feel free to remove sensors/controls which doesn't apply to your hardware.

christo1331, 2025-06-24:
thank you very much for your help , you did it so fast!

i will test it tomorow, it was my first attempt to test it ,sure i will remove everything i don’t use.

good evening.