syssi/esphome-jk-bms #792: Support for JK-BD6A24S10P with Hardware Ver V15A and Software Ver V15.29

Mofazzul, 2025-07-25:
I’m trying to integrate my JK-BD6A24S10P BMS (Hardware Ver: V15A, Software Ver: V15.29) with ESPHome using the esphome-jk-bms component via BLE, but I’m encountering issues. The BMS connects and sends a "Cell info frame (JK04, 300 bytes)," but all sensor values (e.g., total voltage, current, SoC) are reported as 0.00000. The JK-BMS app shows valid data

[image]

, confirming the BMS is functional.

Config File:

Here’s my current jk-bms.yaml:
text
esphome:
  name: jk-bms
  friendly_name: "JK BMS Monitor"
  platform: ESP32
  board: nodemcu-32s

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  ap:
    ssid: "jk-bms-fallback"
    password: !secret ota_password

logger:
  level: DEBUG

api:
  encryption:
    key: !secret api_encryption_key

ota:
  platform: esphome
  password: !secret ota_password

web_server:
  port: 80

captive_portal:

esp32_ble_tracker:
  scan_parameters:
    active: true
    interval: 600ms
    window: 500ms

external_components:
  - source:
      type: git
      url: https://github.com/syssi/esphome-jk-bms
      ref: ff022292c9ed289788221792997dfb545b822302
    refresh: 0s

ble_client:
  - mac_address: "C8:47:80:1E:33:3F"
    id: client0

jk_bms_ble:
  - ble_client_id: client0
    id: bms0
    protocol_version: JK04

sensor:
  - platform: jk_bms_ble
    jk_bms_ble_id: bms0
    total_voltage:
      name: "JK BMS Total Voltage"
    current:
      name: "JK BMS Current"
    power:
      name: "JK BMS Power"
    charging_power:
      name: "JK BMS Charging Power"
    discharging_power:
      name: "JK BMS Discharging Power"
    state_of_charge:
      name: "JK BMS SoC"
    average_cell_voltage:
      name: "JK BMS Average Cell Voltage"
    max_cell_voltage:
      name: "JK BMS Max Cell Voltage"
    min_cell_voltage:
      name: "JK BMS Min Cell Voltage"
    temperature_sensor_1:
      name: "JK BMS Temp Sensor 1"
    temperature_sensor_2:
      name: "JK BMS Temp Sensor 2"
    temperature_sensor_3:
      name: "JK BMS Temp Sensor 3"
    temperature_sensor_4:
      name: "JK BMS Temp Sensor 4"
    temperature_sensor_5:
      name: "JK BMS Temp Sensor 5"
    balancing_current:
      name: "JK BMS Balancing Current"

Logs:

Here are the relevant logs:
text
[01:31:07][C][wifi:599]: WiFi:
[01:31:07][C][wifi:427]:   Local MAC: 00:4B:12:2E:A1:A8
[01:31:07][C][wifi:432]:   SSID: 'Mofazzul's Home 2.4G'
[01:31:07][C][wifi:435]:   IP Address: 192.168.8.23
[01:31:07][C][wifi:439]:   BSSID: A2:83:0C:B4:06:0B
[01:31:07][C][wifi:440]:   Hostname: 'jk-bms'
[01:31:07][C][wifi:442]:   Signal strength: -73 dB ▂▄▆█
[01:31:07][C][wifi:446]:   Channel: 1
[01:31:07][C][wifi:447]:   Subnet: 255.255.255.0
[01:31:07][C][wifi:448]:   Gateway: 192.168.8.1
[01:31:07][C][wifi:449]:   DNS1: 192.168.8.1
[01:31:07][C][wifi:450]:   DNS2: 0.0.0.0
[01:31:07][D][esp32_ble_client:110]: [0] [C8:47:80:1E:33:3F] ESP_GATTC_NOTIFY_EVT
[01:31:07][I][jk_bms_ble:680]: Cell info frame (JK04, 300 bytes) received
[01:31:07][D][sensor:094]: 'JK BMS Min Cell Voltage': Sending state 0.00000 V with 3 decimals of accuracy
[01:31:07][D][sensor:094]: 'JK BMS Max Cell Voltage': Sending state 0.00000 V with 3 decimals of accuracy
[01:31:07][D][sensor:094]: 'JK BMS Average Cell Voltage': Sending state 0.00000 V with 3 decimals of accuracy
[01:31:07][D][sensor:094]: 'JK BMS Total Voltage': Sending state 0.00000 V with 3 decimals of accuracy
[01:31:07][D][jk_bms_ble:798]: Unknown210: 0x00 0x00 0x00 0x00 (always 0x00 0x00 0x00 0x00)
[01:31:07][D][jk_bms_ble:802]: Unknown214: 0x3F 0x00 0x01 0x00 (0xFF 0xFF: 24 cells?)
[01:31:07][D][jk_bms_ble:805]: Unknown218: 0x00
[01:31:07][D][jk_bms_ble:808]: Unknown219: 0x00
[01:31:07][D][jk_bms_ble:816]: Unknown221: 0x03
[01:31:07][D][sensor:094]: 'JK BMS Balancing Current': Sending state 0.00000 A with 2 decimals of accuracy
[01:31:07][D][jk_bms_ble:823]: Unknown226: 0x3C 0xA9 0x3F 0x40 0xBB 0x00 0x00 (always 0x00...0x00?)
[01:31:07][D][jk_bms_ble:827]: Unknown233: 0x00 0x6B 0x14 0x00 (0.000000)
[01:31:07][D][jk_bms_ble:831]: Unknown237: 0x00 0x00 0x00 0x00 (always 0x40 0x00 0x00 0x00?)
[01:31:07][D][jk_bms_ble:845]: Unknown290: 0xDC 0x0F 0x01 0x00 (always 0x00 0x00 0x00 0x00?)
[01:31:07][D][jk_bms_ble:848]: Unknown294: 0x80 0x07 0x00 0x00
[01:31:07][D][jk_bms_ble:851]: Unknown298: 0x00
[01:31:07][D][esp32_ble_client:110]: [0] [C8:47:80:1E:33:3F] ESP_GATTC_NOTIFY_EVT
[01:31:07][D][esp32_ble_client:110]: [0] [C8:47:80:1E:33:3F] ESP_GATTC_NOTIFY_EVT
[01:31:07][W][component:237]: Component esp32_ble took a long time for an operation (163 ms).
[01:31:07][W][component:238]: Components should block for at most 30 ms.

Steps to Reproduce:

    Configure ESP32 with the provided jk-bms.yaml.
    Upload firmware via OTA to IP 192.168.8.23.
    Monitor logs and observe zero sensor values despite a valid JK04 frame.

Expected Behavior:

Sensor values (e.g., total voltage, current, SoC) should match the JK-BMS app readings (e.g., 24.5V, 75%).

Actual Behavior:

All sensor values are 0.00000, even though the app shows valid data.

Additional Context:

    ESPHome version: 2024.6.0
    ESP32 board: nodemcu-32s
    The component commit ff022292c9ed289788221792997dfb545b822302 was used, which added discharge precharge time support, but it didn’t resolve the issue.
    Signal strength is -73 dB, which should be sufficient.
    I’ve tried adjusting BLE scan intervals (600ms/500ms) without success.

Request:

Could support for V15.29 firmware be added or investigated? I’m happy to provide more logs or assist with testing if needed.

Mofazzul, 2025-07-26:
Thank you very much! It works now with that protocol you said..

closed 2025-07-26 by Mofazzul

syssi, 2025-07-26:
Have fun!