syssi/esphome-jk-bms #798: JK-B2A8S-20P unable to retrieve some BMS data after ESPHome update from 2025.7.3 to 2025.7.5

mrmaximas, 2025-08-05:
```
19:21:10	[D]	[button:010]	
'jk-bms-msq retrieve settings' Pressed.
19:21:10	[D]	[jk_bms_ble:1524]	
Write register: AA.55.90.EB.96.00.00.00.00.00.00.00.00.00.00.00.00.00.00.10 (20)
19:21:10	[W]	[jk_bms_ble:314]	
CRC check failed! 0x96 != 0x35
```
some parameters are no longer displayed
[image]
[image]
BMS Info
[image]

syssi, 2025-08-05:
Did you update the BMS firmware recently too? Could you provide an ESPHome log with log level `VERY_VERBOSE` enabled? I would like to see the raw traffic (especially the raw settings frame) of your hardware/software version.

mrmaximas, 2025-08-05:
> Did you update the BMS firmware recently too?

I haven't updated the BMS firmware, it's a little over a week old, it's like that out of the box, moreover, I don't know how to update it ;) right now the data was being readed by esp, but the retrieve settings button still causes the CRC error

syssi, 2025-08-05:
Thanks! This is a complete settings frame (raw) of your log:

```
0x55, 0xAA, 0xEB, 0x90, 0x01, 0xA6, 0xAC, 0x0D, 0x00, 0x00, 0xF6, 0x09, 0x00, 0x00, 0x54, 0x0B, 0x00, 0x00, 0x42, 0x0E, 0x00, 0x00, 0x48, 0x0D, 0x00, 0x00, 0x05, 0x00, 0x00, 0x00, 0x52, 0x0D, 0x00, 0x00, 0x8C, 0x0A, 0x00, 0x00, 0x8E, 0x0D, 0x00, 0x00, 0x16, 0x0D, 0x00, 0x00, 0xBA, 0x09, 0x00, 0x00, 0x50, 0xC3, 0x00, 0x00, 0x3C, 0x00, 0x00, 0x00, 0x3C, 0x00, 0x00, 0x00, 0xA0, 0x86, 0x01, 0x00, 0x2C, 0x01, 0x00, 0x00, 0x3C, 0x00, 0x00, 0x00, 0x1E, 0x00, 0x00, 0x00, 0xD0, 0x07, 0x00, 0x00, 0xBC, 0x02, 0x00, 0x00, 0x58, 0x02, 0x00, 0x00, 0xBC, 0x02, 0x00, 0x00, 0x58, 0x02, 0x00, 0x00, 0x9C, 0xFF, 0xFF, 0xFF, 0x00, 0x00, 0x00, 0x00, 0x20, 0x03, 0x00, 0x00, 0xBC, 0x02, 0x00, 0x00, 0x08, 0x00, 0x00, 0x00, 0x01, 0x00, 0x00, 0x00, 0x01, 0x00, 0x00, 0x00, 0x01, 0x00, 0x00, 0x00, 0xA0, 0x86, 0x01, 0x00, 0x05, 0x00, 0x00, 0x00, 0xF0, 0x0A, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x01, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x20, 0xA1, 0x07, 0x00, 0x80, 0x10, 0xF6, 0x00, 0x18, 0xFE, 0xFF, 0xFF, 0xFF, 0x1F, 0xA9, 0x6F, 0x02, 0x00, 0x00, 0x00, 0x00, 0xDE, 0x55, 0xAA, 0xEB, 0x90, 0x03, 0xA5, 0x4A, 0x4B, 0x5F, 0x42, 0x32, 0x41, 0x38, 0x53, 0x32, 0x30, 0x50, 0x00, 0x00, 0x00, 0x00, 0x00, 0x31, 0x39, 0x48, 0x00, 0x00, 0x00, 0x00, 0x00, 0x31, 0x39, 0x2E, 0x30, 0x35, 0x00, 0x00, 0x00, 0x78, 0x90, 0x06, 0x00, 0x03, 0x00, 0x00, 0x00, 0x35, 0x30, 0x34, 0x31, 0x38, 0x35, 0x43, 0x32, 0x43, 0x31, 0x30, 0x31, 0x34, 0x32, 0x35, 0x00, 0x31, 0x32, 0x33, 0x33, 0x32, 0x31, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x32, 0x35, 0x30, 0x37, 0x33, 0x31
```

Let's verify the CRC to get an idea the implementation is able to handle the frame in general.

syssi, 2025-08-05:
Wow. It looks like you are on the right track. The `ble_client` component of ESPHome was changed recently and may be a regression was introduced. Could you create an issue at the ESPHome project and add a reference here?

mrmaximas, 2025-08-06:
@Macrisu thanks, it's for JK-PBx series, but i have JK-Bx instead.

syssi, 2025-08-25:
@mrmaximas Is this issue resolved?

mrmaximas, 2025-08-25:
@syssi no issue observed after last ESPHome 2025.8.0 update 🙏🏻

syssi, 2025-08-25:
Cool. Alright!