syssi/esphome-jk-bms #806: problems connecting the second BMS

Vladd1980, 2025-08-20:
good day! I have been using this project for several months and it worked great. But today I turned on the second battery, respectively the second bms B1A8S10P. After that, something unexpected happened. As soon as bms1 is turned on, the bluetooth connection becomes unstable, constantly loses connection with both bms and in fact nothing works. As soon as I turn off bms1 - everything returns to its normal state. 
I will post my configuration file for esp32 here, maybe someone can help me.

substitutions:
  name: jk-bms
  friendly_name: "JK BMS"
  bms0: "${name} bms0"
  bms1: "${name} bms1"
  device_description: "Мониторинг и управление JK-BMS через bluetooth"
  bms0_mac_address: "C8:47:80:06:0A:53"
  bms0_protocol_version: "JK02_32S"
  bms1_mac_address: "C8:47:80:3E:B9:5F" 
  bms1_protocol_version: "JK02_32S"
  tx_pin: GPIO16
  rx_pin: GPIO17

external_components:
  - source: github://syssi/esphome-jk-bms@main

esphome:
  name: ${name}
  comment: ${device_description}
  min_version: 2024.6.0
  project:
    name: "syssi.esphome-jk-bms"
    version: 2.1.0

esp32:
  board: esp32dev
  framework:
    type: esp-idf

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  use_address: 192.168.203.92

web_server:
  port: 80

ota:
  platform: esphome
  password: "f36339b11c2585254cdb311f444e67f3"
  on_begin:
    then:
      - switch.turn_off: ble_client_switch0
      - switch.turn_off: ble_client_switch1
      - logger.log: "BLE-соединение приостановлено для обновления OTA"

logger:
  level: VERY_VERBOSE
  logs:
    esp32_ble_tracker: INFO
    esp32_ble_client: INFO

# Если вы не используете Home Assistant, удалите этот раздел `api` и раскомментируйте компонент `mqtt`!
api:
  encryption:
    key: "469SyXgknWj2YrbvuwbuV/hnHiMtSLGJPDW2t9duNHQ="

# mqtt:
#   broker: !secret mqtt_host
#   username: !secret mqtt_username
#   password: !secret mqtt_password
#   id: mqtt_client

esp32_ble_tracker:
  scan_parameters:
    active: false

[... 403 lines cut ...]

  - platform: ble_client
    ble_client_id: client0
    id: ble_client_switch0
    name: "${bms0} enable bluetooth connection"

  - platform: ble_client
    ble_client_id: client1
    id: ble_client_switch1
    name: "${bms1} enable bluetooth connection"
### Text Sensors

text_sensor:
  - platform: jk_bms_ble
    jk_bms_ble_id: bms0
    errors:
      name: "${bms0} errors"
    total_runtime_formatted:
      name: "${bms0} total runtime formatted"
  - platform: jk_bms_ble
    jk_bms_ble_id: bms1
    errors:
      name: "${bms1} errors"
    total_runtime_formatted:
      name: "${bms1} total runtime formatted"

syssi, 2025-08-21:
Which ESPHome version do you use?

syssi, 2025-08-21:
Alright. Let's close this issue.

closed 2025-08-21 by syssi