syssi/esphome-jk-bms #807: esphome 2025.8    [W][jk_modbus:071]: Invalid header: 0x4E 0x53

dwt15, 2025-08-22:
after the release of update 2025.8, here is the error [W][jk_modbus:071]: Invalid header: 0x4E 0x53

dwt15, 2025-08-22:
rolled back to 2025.7.5 everything works

syssi, 2025-08-22:
Could you provide your YAML configuration? Do you use a ESP8266 or ESP32?

syssi, 2025-08-22:
The "invalid header" issue could be an race condition too. If you reboot the ESP32  (using 2025.8.0) multiple times goes the "invalid header" issue away?

syssi, 2025-08-22:
Unfortunately I'm not able to reproduce the issue:

```
[11:31:31][D][debug:033]: ESPHome version 2025.8.0
[11:31:31][D][debug:037]: Free Heap Size: 76668 bytes
[11:31:31][D][debug:182]: Chip: Model=ESP32, Features=2.4GHz WiFi, BLE, BT,  Cores=2, Revision=100
[11:31:31][D][debug:191]: CPU Frequency: 160 MHz
[11:31:31][D][debug:199]: Framework: ESP-IDF
[11:31:31][D][debug:206]: ESP-IDF Version: 5.4.2
[11:31:31][D][debug:211]: EFuse MAC: 7C:9E:BD:06:B5:58
[11:31:31][D][debug:079]: Reset Reason: Reboot request from esphome.ota
[11:31:31][D][debug:107]: Wakeup Reason: undefined
[11:31:31][C][debug:112]: Partition table:
[11:31:31][C][debug:112]:   Name         Type Subtype  Address    Size      
[11:31:31][C][debug:119]:   nvs          1    2        0x00009000 0x00004000
[11:31:31][C][debug:119]:   otadata      1    0        0x0000D000 0x00002000
[11:31:31][C][debug:119]:   phy_init     1    1        0x0000F000 0x00001000
[11:31:31][C][debug:119]:   app0         0    16       0x00010000 0x001C0000
[11:31:31][C][debug:119]:   app1         0    17       0x001D0000 0x001C0000
[11:31:32][I][jk_bms:071]: Status frame received
[11:31:37][I][jk_bms:071]: Status frame received
[11:31:42][I][jk_bms:071]: Status frame received
[11:31:47][I][jk_bms:071]: Status frame received
[11:31:52][I][jk_bms:071]: Status frame received
```

dwt15, 2025-08-22:
Sorry for being persistent but where is the uart
on version: 5.4.1 some sensors came to life but do not get into the assistant

syssi, 2025-08-22:
Could you rephrase your question and statement? My `uart` is attached to GPIO14 and GPIO4:

```
uart:
  id: uart_0
  baud_rate: 115200
  rx_buffer_size: 384
  tx_pin: GPIO14
  rx_pin: GPIO4

jk_modbus:
  id: modbus0
  uart_id: uart_0

jk_bms:
  id: bms0
  jk_modbus_id: modbus0
```

Which sensors did change / come to life? Did I understand your diagnosis correctly: ESP-IDF framework 5.4.2 is affected and 5.4.1 not?

dwt15, 2025-08-22:
cheap yellow display with pins rx0 and th0 problems i used them the other way around

syssi, 2025-08-22:
How did you solve the issue?

dwt15, 2025-08-22:
decided to use other pins

dwt15, 2025-08-22:
[image]

[image]

syssi, 2025-08-22:
Do you like to contribute the YAML code of your display?

dwt15, 2025-08-22:
[batteryx.yaml](https://github.com/user-attachments/files/21945087/batteryx.yaml)